peplin/pygatt #23: VersionConflict with nose on install

cadavor, 2015-08-27:
Hello,

When i try to install your script, I get this error : 
pkg_resources.VersionConflict: (nose 1.3.7 (/usr/local/lib/python2.7/dist-packages), Requirement.parse('nose==1.3.1'))
Can you change your requirement ?

peplin, 2015-08-27:
Hi there - you may want to use a [virtualenv](https://virtualenv.pypa.io/en/latest/) so the version of `nose` installed globally on your system doesn't conflict. We will probably update our nose version at some point, but in the meantime it shouldn't block you from installing.

cadavor, 2015-08-27:
I'm on Rpi2 with Raspbian

Here the log : 
Downloading/unpacking pygatt[gatttool]
  Running setup.py egg_info for package pygatt
    Traceback (most recent call last):
      File "<string>", line 14, in <module>
      File "/home/jeedom/build/pygatt/setup.py", line 54, in <module>
        include_package_data=True
      File "/usr/lib/python2.7/distutils/core.py", line 112, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 221, in __init__
        self.fetch_build_eggs(attrs.pop('setup_requires'))
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 245, in fetch_build_eggs
        parse_requirements(requires), installer=self.fetch_build_egg
      File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 588, in resolve
        raise VersionConflict(dist,req) # XXX put more info here
    pkg_resources.VersionConflict: (nose 1.3.7 (/usr/local/lib/python2.7/dist-packages), Requirement.parse('nose==1.3.1'))
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 14, in <module>

  File "/home/jeedom/build/pygatt/setup.py", line 54, in <module>

    include_package_data=True

  File "/usr/lib/python2.7/distutils/core.py", line 112, in setup

    _setup_distribution = dist = klass(attrs)

  File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 221, in __init__

    self.fetch_build_eggs(attrs.pop('setup_requires'))

  File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 245, in fetch_build_eggs

    parse_requirements(requires), installer=self.fetch_build_egg

  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 588, in resolve

    raise VersionConflict(dist,req) # XXX put more info here

pkg_resources.VersionConflict: (nose 1.3.7 (/usr/local/lib/python2.7/dist-packages), Requirement.parse('nose==1.3.1'))

----------------------------------------
Command python setup.py egg_info failed with error code 1 in /home/jeedom/build/pygatt
Storing complete log in /home/jeedom/.pip/pip.log

Thanks for your help...

peplin, 2015-08-27:
You may want to try out the `next` branch of this repository, I remember now we've actually dropped `nose` from the install dependencies.

cadavor, 2015-08-27:
Sorry to ask that but how to use the next branch? (noob inside)

stratosqa, 2015-08-27:
No problem, run "git checkout next" to switch branches

cadavor, 2015-08-27:
Sorry, but after doing : 
pip install https://github.com/stratosinc/pygatt
I got that : 
pip install https://github.com/stratosinc/pygatt
Downloading/unpacking https://github.com/stratosinc/pygatt
  Downloading pygatt (unknown size): 55Kb downloaded
  Cannot unpack file /tmp/pip-ryy7Dt-unpack/pygatt (downloaded from /tmp/pip-x_XHGC-build, content-type: text/html; charset=utf-8); cannot detect archive format
Cannot determine archive format of /tmp/pip-x_XHGC-build
Storing complete log in /home/jeedom/.pip/pip.log

I think i've got a problem firstly here no?

Thanks for your help

stratosqa, 2015-08-27:
Try this:

pip install "git+https://github.com/stratosinc/pygatt@next"

cadavor, 2015-08-27:
Do not take care about my previous message (if you've read it)...

To solve the problem, I've forced installation of nose 1.3.1. (sudo pip install nose==1.3.1)
And coverage 3.7.1 and the installation complete...

peplin, 2015-08-27:
OK, great!

closed 2015-08-27 by peplin

cadavor, 2015-08-28:
Thanks for your help.