peplin/pygatt #36: pip install broken

IlyaSukhanov, 2016-01-13:
Upon installing pygatt via pip yields in broken package. Import fails with following error:
> &gt;&gt;&gt; import pygatt.backends
> Traceback (most recent call last):
>  File "<stdin>", line 1, in <module>
>  File "/home/user.virtualenvs/pexpect3_test/lib/python3.4/site-packages/pygatt/__init__.py", line 3, in > <module>
>    from .backends import BGAPIBackend, GATTToolBackend  # noqa
> ImportError: No module named 'pygatt.backends'

Appears that pygatt.backends is not part of pip distribution. Missing from setup.py perhaps?

pip install -e . against source works without problems.

IlyaSukhanov, 2016-01-14:
opened PR with fix https://github.com/peplin/pygatt/pull/38 before which ```python setup.py sdist``` would generate egg without pygatt-2.0.0/pygatt/backends after, with.

IlyaSukhanov, 2016-02-02:
Christopher do you have access to upload this change to https://pypi.python.org/pypi ? It's somewhat unfortunate that one cannot fetch working version of pygatt from there.

peplin, 2016-02-02:
Yes, I do, I'm planning to get that taken care of soon.

peplin, 2016-02-02:
v2.0.1 released to PyPI, sorry for the delay!

```
>>> import pygatt.backends
>>> 
```

closed 2016-02-02 by peplin