peplin/pygatt #37: Trying to use pygatt on RPi (Jessie)

keptenkurk, 2016-01-13:
Trying to use Pygatt on a Raspberry Pi with latest Rapbian Jessie installed.
Bluez installed Version: 5.23-2+b1
gatttool works and is able to connect to BLE device
latest pexpect installed and working

The script: (with correct address)

    # pygatt test
    import pygatt.backends
    
    adapter = pygatt.backends.GATTToolBackend()
    device = adapter.connect('f1:37:57:XX:XX:XX')
    value = device.char_read("a1e8f5b1-696b-4e4c-87c6-69dfe0b0093b")

returns

    Traceback (most recent call last):
      File "test.py", line 4, in <module>
        device = adapter.connect('f1:37:57:XX:XX:XX')
      File "/home/pi/Pygatt-master/pygatt/backends/gatttool/gatttool.py", line 178, in connect
        self._con.sendline('sec-level low')
    AttributeError: 'NoneType' object has no attribute 'sendline'

Any pointers where to start looking? 
/paul

IlyaSukhanov, 2016-01-14:
Try adapter.start() before adapter.connect(). Its a bit convoluted, seems to me like start() should be called part of __init__(). Not sure why this is the case. Maybe docs should be updated if this is in fact desired behavior.

keptenkurk, 2016-01-14:
Thanks Ilya! That did the trick!

closed 2016-01-14 by keptenkurk

peplin, 2016-01-14:
Thanks @IlyaSukhanov - I created #39 since that's a mistake leftover from a refactor.