peplin/pygatt #39: Start gatttool backend as soon as it is initialized

peplin, 2016-01-14:
In #37, @IlyaSukhanov mentioned that this backend has a `start()` method that has to be called before it can be used, but that isn't mentioned anywhere. This function is a remnant of the v1.0 API, where each backend had a start method. It should now handle that on its own, since the BGAPI backend does not need this.

peplin, 2016-01-14:
Actually my first comment here was incorrect - the BGAPI *does* have a start method and the docstring does say:

```
Initialize the backend, but don't start the USB connection yet. Must     
        call .start().    
```

so I think my intent was to separate initialization and actually starting, but I forgot to mention this in the docs. I'll update the title.

keptenkurk, 2016-01-14:
And maybe add a link to [this blog](http://flowcloud.github.io/ci20-bluetooth-LE/2015/09/10/bluetooth-control-in-python/) which explains the pygatt workings at newbee level.

commit 29b948b4 (2016-09-04): Update examples in docs and add complete sample program.

Fixes #39
PR #52 (2016-09-04): Update examples in docs and add complete sample program.
Fixes #39
closed 2016-09-04 by peplin