peplin/pygatt #46: gatttool high CPU usage on 'disconnect' command

gkopendev, 2016-03-29:
Hi,
This is not exactly a pygatt issue, but I just wanted to get some inputs. Doing a gatttool disconnect causes CPU usage of 70-80% till gatttool is killed. I am using bluez 5.37. I also notice this issue has been noted as a TODO in gatttool.py. So has anyone had a go at trying to resolve this?

Thanks

gkopendev, 2016-04-02:
If anyone looks for this I describe the solution here - 
https://communities.intel.com/thread/100451
http://stackoverflow.com/questions/36289272/gatttool-disconnection-in-bluez-5-3-causes-very-high-cpu-usage-linux-bluetooth/36310040#36310040

peplin, 2016-09-04:
I am not seeing this issue anymore - it may have been fixed upstream.

closed 2016-09-04 by peplin