peplin/pygatt #49: Unable to auto-detect BLED112 serial port

amahlaka, 2016-07-04:
Hi, im trying to use pygatt on raspi, when i run ``` adapter = pygatt.backends.BGAPIBackend() ``` i get 
``` Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.4/dist-packages/pygatt/backends/bgapi/bgapi.py", line 79, in __init__
    raise BGAPIError("Unable to auto-detect BLED112 serial port") ```
what does this mean?
i have a usb bluetooth 4 le dongle connected
im running python as sudo

amahlaka, 2016-07-04:
Fixed using ```  adapter = pygatt.backends.GATTToolBackend() ```

closed 2016-07-04 by amahlaka

peplin, 2016-07-07:
Really? I'm surprised by that - the BLED112 doesn't work with gatttool/BlueZ as far as I know.

weilonluo, 2019-07-20:
> Fixed using adapter = pygatt.backends.GATTToolBackend()
BLEError: The GATTToolBackend requires BlueZ, which is not available in Windows

goodtalen, 2019-11-25:
self.adapter = pygatt.backends.GATTToolBackend()

   raise BLEError("The GATTToolBackend requires BlueZ, "
pygatt.exceptions.BLEError: The GATTToolBackend requires BlueZ, which is not available in Windows

amahlaka, 2019-11-25:
As the error says, blueZ is not available in windows

ErfolgreichCharismatisch, 2021-12-30:
so pygatt cannot be used under windows at all?

chadkennedyonline, 2022-01-04:
Under Windows, pygatt works great, using the BLED112 adapter, along with the BGAPI backend.

ErfolgreichCharismatisch, 2022-01-04:
I don't have the bled112 adapter. How do I use it under Windows?

chadkennedyonline, 2022-01-04:
As far as I know, you don't. You'd have to buy a BLED112. I checked and they're available on mouser.de for 19,77 €: https://www.mouser.de/ProductDetail/Silicon-Labs/BLED112-V1?qs=2bnkQPT%252BPaf%252BBQ6Sw8mJqQ%3D%3D