peplin/pygatt #50: Writing to handle

amahlaka, 2016-07-04:
Hi, i tried everyhting i could but i cant get device.char_write_handle() to work.
im trying to write to handle 0x0025 the value ff000000, i havent been able to find any propper documentation on how to use this function so i cant seem to get it to work
@peplin

peplin, 2016-07-07:
What type of device are you using, and can you paste the code you are using to connect? I saw in #49 that you have a BLED112 but you suggest that you are trying to use the GATTTool backend - those are not compatible, so that's likely why you can't get writing to handles to work.

peplin, 2016-09-04:
I'm going to close this one due to inactivity - I just tested writing and it's working OK. Please re-open if you are still having trouble. The code should look something like:

```python
device.char_write(uuid, bytearray([1, 2]))
```

This would write `0x01 0x02` to the UUID in `uuid`.

closed 2016-09-04 by peplin