peplin/pygatt #66: Installing BGAPI version to windows

amahlaka, 2016-09-25:
is it possible to install pygatt to windows with the BGAPI
i understand that the bluez wont work as its linux only, but how about bgapi?

peplin, 2016-09-26:
The BGAPI backend should work in Windows - in fact this was the primary reason for adding this backend. If you install the software for the BLED112 dongle (or whatever other BGAPI compatible device you have), pygatt _should_ just work. I haven't tested it on Windows in a few months though, so if you have issues please open a new ticket to let me know.

amahlaka, 2016-09-26:
The problem im having is:   File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\pygatt\__init__.py", line 3, in <module>
    from .backends import BGAPIBackend, GATTToolBackend, BLEAddressType  # noqa
  File "C:\Python27\lib\site-packages\pygatt\backends\__init__.py", line 2, in <module>
    from .bgapi.bgapi import BGAPIBackend  # noqa
  File "C:\Python27\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 10, in <module>
    import termios
ImportError: No module named termios
>>>

peplin, 2016-09-26:
Ahh, thanks, this is a regression from 17481f5d187eeb9b0aeab04d94616e3e2d0cb347 then.

amahlaka, 2016-09-26:
is there any way to access the BT LE on windows 10

peplin, 2016-09-27:
If you want to try removing the references to `termios` in pygatt, that would be a good test. If that works for you, then we need a way to gracefully ignore when that library is not available (it's only in Cygwin, I think).

tmecha, 2016-10-24:
I just had this same issue on Windows trying to connect to BLED112

commit dcf8deb7 (2016-11-19): Support platforms without termios import (e.g. Windows)

Fixes #66
PR #76 (2016-11-19): Support platforms without termios import (e.g. Windows)
Fixes #66
closed 2016-11-19 by peplin