peplin/pygatt #67: BGAPIError: Unable to auto-detect BLED112 serial port

mettinger, 2016-09-26:
On a Mac after:

import pygatt
adapter = pygatt.BGAPIBackend()
adapter.start()

results in this error:
---------------------------------------------------------------------------
BGAPIError                                Traceback (most recent call last)
<ipython-input-3-9b89dcd34ca5> in <module>()
----> 1 adapter.start()

/Users/mark/anaconda2/lib/python2.7/site-packages/pygatt/backends/bgapi/bgapi.pyc in start(self)
    162             self.stop()
    163
--> 164         self._open_serial_port()
    165
    166         # Blow everything away and start anew.

/Users/mark/anaconda2/lib/python2.7/site-packages/pygatt/backends/bgapi/bgapi.pyc in _open_serial_port(self)
    134         attempts, with a short pause in between each attempt.
    135         """
--> 136         serial_port = self._serial_port or self._detect_device_port()
    137         self._ser = None
    138         for _ in range(MAX_RECONNECTION_ATTEMPTS):

/Users/mark/anaconda2/lib/python2.7/site-packages/pygatt/backends/bgapi/bgapi.pyc in _detect_device_port(self)
    120             product_id=BLED112_PRODUCT_ID)
    121         if len(detected_devices) == 0:
--> 122             raise BGAPIError("Unable to auto-detect BLED112 serial port")
    123
    124         return detected_devices[0].port_name

BGAPIError: Unable to auto-detect BLED112 serial port

peplin, 2016-09-27:
I'm not too familiar with OS X , but can you look at the enumerated USB devices when you plug the dongle in? It might be in the System Profiler, if that's a thing.

 The auto-detection is looking for a USB device with this vendor and product ID:

```
BLED112_VENDOR_ID = 0x2458                                                       
BLED112_PRODUCT_ID = 0x0001 
```

Does one of those devices appear? Something else with the same vendor but a different product ID, perhaps?

mettinger, 2016-09-28:
I'm not using a dongle.  My Mac supports BLE so I assumed this would be sufficient.  I have no previous experience with bluetooth so perhaps this is foolish.

peplin, 2016-09-29:
I understand now, thanks - the BGAPI backend is meant to support USB dongles made by Bluegiga, like the [BLED112](http://www.silabs.com/products/wireless/bluetooth/bluetooth-smart-modules/Pages/bled112-bluetooth-smart-dongle.aspx).

The other supported backend in pygatt at the moment is based on BlueZ, which is only available in Linux.

So, unfortunately for Mac OS X you either need to buy a BLED112 or look elsewhere :/

closed 2016-09-29 by peplin

amias-channer, 2017-05-04:
I have a BLED112 on OSX Sierra , it shows the device is connected but pygatt says 'unable to open adapter'

if I run "system_profiler SPUSBDataType" , amongst many other things , I get 

            Low Energy Dongle:

              Product ID: 0x0001
              Vendor ID: 0x2458
              Version: 0.01
              Serial Number: 1
              Speed: Up to 12 Mb/sec
              Manufacturer: Bluegiga
              Location ID: 0x14520000 / 14
              Current Available (mA): 500
              Current Required (mA): 74
              Extra Operating Current (mA): 0

so I'm confident its setup ok , trying to connect with pygatt gives me 

ERROR:root:Unable to open bluetooth adapter
ERROR:root:No adapter available - quitting

Is there a way to get more info or verify the dongle (e.g. connecting to it serial port and observing something) ?