peplin/pygatt #68: NotificationTimeout exception and NotConnectedError while trying to read battery value of sensor.

BhargavaRamM, 2016-10-07:
Hey.. 
I am trying to connect to a BLE device from a raspberry pi and using pygatt for connecting, reading and writing values from and to sensor to get notifications. Most of the times I get what I want but some times I receive **NotificationTimeout** and **NotConnectedError**(even though the device is connected).  
I am receiving these errors while trying to read the battery value of a bluetooth sensor using **char_read** function. Also, these errors are sporadic and I can not pin point when it is going to happen. Any help is appreciated. Thank you.

peplin, 2016-10-07:
BLE devices in my experience tend to be fairly flaky when it comes to receiving data - they are constantly trying to turn off and go into low power modes, hence the "low energy" part of the name. I usually find that you have to make your code robust to occasional timeouts.

You may be getting a NotConnectedError because while waiting for the notification, the device actually disconnects from your raspberry pi.

BhargavaRamM, 2016-10-08:
@peplin Thank you. I installed pygatt using pip. Is there a way I can extend the notification timeout period?

peplin, 2016-10-09:
There is no timeout for notifications, since the idea is that you wait as long as the device is connected to receive the notification. The "NotificationTimeout" is actually an internal detail of the library, and not referring to BLE notifications. I believe your device is actually disconnecting, so waiting longer for a notification will not help.

If you want to try experimenting with `gatttool`'s interactive mode, that might give you a sense of what's going on - try opening a connection and see how long you can keep it open before the device forces a disconnection.

BhargavaRamM, 2016-10-18:
@peplin Thank you. I played with gatttool before I became aware of pygatt. But, developing an application using Bluez required DBUS. Pygatt made my life easy. Anyway, you can go ahead and close this issue.

closed 2016-10-23 by BhargavaRamM