peplin/pygatt #69: ExceptionPexpect: The command was not found or was not executable: gatttool.

Alionea, 2016-10-11:
Hello,

i use a new Raspberry pi3 with the newest Version of Raspbian. I followed every instruction to install Bluez, Gatttool and pygatt.  I have no Idea how to manage this Error. Please can you help me.
The  Error is:

Traceback (most recent call last):
  File "/home/pi/MIA/char_read.py", line 12, in <module>
    adapter.start()
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 222, in start
    self._con = pexpect.spawn(gatttool_cmd, logfile=self._gatttool_logfile)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 197, in __init__
    self._spawn(command, args, preexec_fn, dimensions)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 270, in _spawn
    'executable: %s.' % self.command)
ExceptionPexpect: The command was not found or was not executable: gatttool.

Thank you!

Alionea, 2016-10-11:
Ok i found the Error- writing only

device = adapter.connect("D6:2C:B6:38:DF:24") 

but why? I have the same skript running on a different image (instatlled jessy lite)- and there it works fine...

peplin, 2016-10-11:
Does your device use public or random addressing? The `connect` function requires an `pygatt.BLEAddressType` be passed in to the `address_type` argument - from the first error you posted, it looks like you are trying to pass a string. This changed recently with the latest version of Pygatt, which probably explains why the script only fails on the new board (see the [CHANGELOG](https://github.com/peplin/pygatt/blob/master/CHANGELOG.mkd) for details).

If you do need random addressing, change your script to say:

```
ADDRESS_TYPE = pygatt.BLEAddressType.random
device = adapter.connect("D6:2C:B6:38:DF:24", address_type=ADDRESS_TYPE)
```

(see the [updated example](https://github.com/peplin/pygatt/blob/master/samples/basic_bluez_gatttool_sample.py))

Please re-open if that doesn't fix your issue!

closed 2016-10-11 by peplin