peplin/pygatt #71: Unable to auto-detect BLED112 after reboot

iadjedj, 2016-10-18:
Hi,
Since last update, auto-detect of a BLED112 fails.
After some investigations, it seems that the error is caused by the reboot of the serial port. Seems like the BLED112 doesn't have enough time to give full informations to the auto-detect.

> INFO:pygatt.backends.bgapi.bgapi:Initialized new BGAPI backend on None
INFO:pygatt.backends.bgapi.bgapi:Auto-discovering serial port for BLED112
INFO:pygatt.backends.bgapi.util:Found 1 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: /dev/ttyACM0 - Low Energy Dongle - CDC control
DEBUG:pygatt.backends.bgapi.util:USB device: Low Energy Dongle - CDC control VID=0x2458 PID=0x0001 on /dev/ttyACM0
DEBUG:pygatt.backends.bgapi.bgapi:Attempting to connect to serial port after restarting device
INFO:pygatt.backends.bgapi.bgapi:Auto-discovering serial port for BLED112
INFO:pygatt.backends.bgapi.util:Found 1 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: /dev/ttyACM0 - n/a
Unable to auto-detect BLED112 serial port

I'm on Ubuntu 16.04, never had this problem before last update.

peplin, 2016-10-25:
At the end of the log, it looks like it actually did detect the device after resetting it, but it may not be fully initialized. Would you be able to try adding a second or two of delay after the reset, before we attempt to detect the device again?

iadjedj, 2016-10-25:
Adding a one second delay after the reset fixes the problem, another fix could be doing the auto-detect in the connection retry loop.
I've tried both solutions and they appears to work fine.

PR #79 (2016-11-19): Retry BGAPI device discover to allow device time to reboot.
Fixes #71
commit 2b6c2e40 (2016-11-19): Retry BGAPI device discover to allow device time to reboot.

Fixes #71
closed 2016-11-19 by peplin