peplin/pygatt #72: Disconnect event not shown.

schef, 2016-10-20:


peplin, 2016-10-23:
Thanks for the report - can you provide more information?

* Which BLE backend?
* Sample code?
* What do you expect to happen? What happens instead?

schef, 2016-10-24:
First of all thank you for great software.
I'm a firmware developer for BLE113 which is pretty hard to test/debug. So i am using python nosetests and pygatt to unittest my firmware code. Nice.

I am on Archlinux and use gatt. The code i have written till now works great. The thing with BLE113 devices is that they often get disconnected. So i need to catch the Disconnect event. I guess that this is implemented but that there is some kind of regex issue and that the callback is not executed.

What i get when i turn off the device (force the disconnection):

`1477293960242: Attempting to connect to 00:07:80:A5:45:A7
1477293964672: Connection successful
[00:07:80:A5:45:A7][LE]> 
`

`1477293960242: Attempting to connect to 00:07:80:A5:45:A7
1477293964672: Connection successful
1477293988962: Disconnected
[00:07:80:A5:45:A7][LE]> 
(gatttool:15093): GLib-WARNING **: Invalid file descriptor.
`

i think that the problem is that the gatttool replaces the line.

NigelTwo, 2016-10-30:
I have also been looking for a disconnect event or a connection status that can be polled. I have a RPi3 communicating to a JNHuaMao HM-10 serial-to-BLE module. Ideally they would stay connected forever. However the RPi acting as the central, needs to re-establish the connection to the peripheral device after any connection breaks. The RPi is running Jessie and I have installed BlueZ 5.42.

Here is the "connect/receive some data/peripheral-disconnect" sequence using gatttool interactively.
First, I call the tool and make the connection. Notice how the mac address changes colour (bolded here) after the successful connection:

pi@NigelPi:~/BLEtest $ sudo gatttool -b D0:B5:C2:9F:8F:32 -I
[D0:B5:C2:9F:8F:32][LE]> connect
Attempting to connect to D0:B5:C2:9F:8F:32
Connection successful
[**D0:B5:C2:9F:8F:32**][LE]>

Now some data is received and is printed as a notification. The coloured mac address remains in the prompt:

pi@NigelPi:~/BLEtest $ sudo gatttool -b D0:B5:C2:9F:8F:32 -I
[D0:B5:C2:9F:8F:32][LE]> connect
Attempting to connect to D0:B5:C2:9F:8F:32
Connection successful
Notification handle = 0x0012 value: 48 65 6c 6c 6f 20 77 6f 72 6c 64
[**D0:B5:C2:9F:8F:32**][LE]>

The peripheral is now turned off. There is no notification in the user interface. However, if the Enter key is pressed, the  prompt is over-written with plain text. This has to be seen to be believed!

pi@NigelPi:~/BLEtest $ sudo gatttool -b D0:B5:C2:9F:8F:32 -I
[D0:B5:C2:9F:8F:32][LE]> connect
Attempting to connect to D0:B5:C2:9F:8F:32
Connection successful
Notification handle = 0x0012 value: 48 65 6c 6c 6f 20 77 6f 72 6c 64
[D0:B5:C2:9F:8F:32][LE]>

So gatttool knows about the disconnection, but requires prompting to display it. There is no specific event. Hence in the following code, the final while loop must either receive an exception, or include some status polling, so that it can be aware of the disconnect.

#!/usr/bin/env python

# Connect to HM-10 BLE in peripheral mode and listen for incoming
# data. Peripheral will disconnect after all data sent.

from __future__ import print_function

import logging
logging.basicConfig()
logging.getLogger('pygatt').setLevel(logging.DEBUG)

import pygatt
import time
import sys

address = "D0:B5:C2:9F:8F:32"

adapter = pygatt.GATTToolBackend()
adapter.start()

while True:
   try:
      print("Connecting...")
      device = adapter.connect(address,3,pygatt.BLEAddressType.public)
      print("Connected")
      break
   except pygatt.exceptions.NotConnectedError:
      print("Not connected")
      time.sleep(1)
      continue

#device.subscribe('0000ffe1-0000-1000-8000-00805f9b34fb', callback = printNotification)

while True:
   try:
      time.sleep(1)
   except Exception as e:
      print(e)

commit 6a80c79a (2016-11-19): Interpret "invalid file descriptor" as a disconnect event.

Fixes #72
PR #78 (2016-11-19): Interpret "invalid file descriptor" as a disconnect event.
This adds the ability for multiple gatttool patterns to trigger the same event. The regex for the disconnect event is also loosened to ignore the line ending.

Fixes #72
commit 02197ff7 (2016-11-19): Interpret "invalid file descriptor" as a disconnect event. (#78)

Fixes #72
closed 2016-11-19 by peplin