peplin/pygatt #74: pexpect.TIMEOUT: Timeout exceeded

nishadsabnis, 2016-11-09:
I am trying to connect to an nRF8001 using pygatt. I am able to connect to it using gatttool but am facing some issues while using pygatt. My code is as follows 
`#!/usr/bin/env python
import binascii
import pygatt
import logging

logging.basicConfig()
logging.getLogger('pygatt').setLevel(logging.DEBUG)

FRONT_MODULE_ADDRESS = "E7:48:F5:CD:DF:D3"
ADDRESS_TYPE = pygatt.BLEAddressType.public

adapter = pygatt.GATTToolBackend();
adapter.start()
device = adapter.connect(FRONT_MODULE_ADDRESS)

adapter.stop()`

I do appear to connect to the peripheral as an LED lights up on it but I get a timeout error from pexpect. 

`DEBUG:pygatt.backends.gatttool.gatttool:gatttool_cmd=gatttool -i hci0 -I
Traceback (most recent call last):
  File "ble.py", line 13, in <module>
    adapter.start()
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 224, in start
    self._con.expect(r'\[LE\]>', timeout=1)

  File "/usr/lib/python2.7/dist-packages/pexpect/__init__.py", line 1418, in expect
    timeout, searchwindowsize)
  File "/usr/lib/python2.7/dist-packages/pexpect/__init__.py", line 1433, in expect_list
    timeout, searchwindowsize)
  File "/usr/lib/python2.7/dist-packages/pexpect/__init__.py", line 1535, in expect_loop
    raise TIMEOUT(str(err) + '\n' + str(self))
pexpect.TIMEOUT: Timeout exceeded.
<pexpect.spawn object at 0xb666fe10>
version: 3.2
command: /usr/bin/gatttool
args: ['/usr/bin/gatttool', '-i', 'hci0', '-I']
searcher: <pexpect.searcher_re object at 0xb666fe50>
buffer (last 100 chars): ''
before (last 100 chars): ''
after: <class 'pexpect.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 26041
child_fd: 3
closed: False
timeout: 30
delimiter: <class 'pexpect.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1` 

I am not really sure if I am doing anything wrong.

nishadsabnis, 2016-11-09:
I should add that when I manually type the "gatttool -i hci0 -I" command on the command line it does return a prompt with space for a blank address and then [LE] so it should not be triggering the pexpect timeout as the regular expression should match. 

Thanks for the library, will definitely be an easier way to connect to BLE peripherals once I get it up and running.

peplin, 2016-11-19:
Which version of BlueZ (i.e. the version of gatttool) do you have? Can you paste the terminal output when you run it manually? There may be some subtle difference in your version.

nishadsabnis, 2016-11-19:
When I run gatttool -I, I get a prompt which looks like this "[                 ][LE]>" 

I am using BlueZ 5.37 which was compiled from the source files. Also, I wrote my own script with pexpect and it seems that the timeout is getting triggered. In gatttool.py you have selected a timeout of 1 for the con expect but if I run my own script with the default timeout for pexpect, it works fine. I guess it is taking more time on my system for some reason! I do not know why others are not facing the same issue.

peplin, 2016-12-28:
Interesting! Roughly how long does it take for the prompt to show up? I can bump the default to something that will cover your case (is 3 seconds enough?) and make it configurable.

closed 2016-12-28 by peplin

nishadsabnis, 2016-12-29:
Thanks for the changes! I do not remember the exact time it was taking, probably a few seconds. But now that it is configurable that's even better. This might be because I was using it on a Pi, not a PC. I do not have access to my Pi right now so cannot give you the exact time it was taking.

commit 31556c4a (2016-12-28): Wait longer for gatttool prompt and make time configurable.

Fixes #74.
PR #90 (2016-12-28): Wait longer for gatttool prompt and make time configurable.
Fixes #74.
commit 340b0068 (2016-12-28): Wait longer for gatttool prompt and make time configurable. (#90)

Fixes #74.