peplin/pygatt #82: BGAPI scan for devices fails because `len(field_value) / 16` cannot be interpreted as an integer

danjjl, 2016-12-08:
A user of a software I am writing is complaining of an occasional crash when he scans for Bluetooth devices.

He reports the crash only occurs occasionally on a macOS 10.12 computer. We are using pygatt  02197ff. I for myself have never experienced this issue on a Ubuntu 16.04 machine.

```
...
  devices = adapter.scan(timeout=1)
File "/usr/local/lib/python3.5/site-packages/pygatt/backends/bgapi/bgapi.py", line 317, in scan
  self.expect(ResponsePacketType.gap_end_procedure)
File "/usr/local/lib/python3.5/site-packages/pygatt/backends/bgapi/bgapi.py", line 517, in expect
  return self.expect_any([expected], *args, **kargs)
File "/usr/local/lib/python3.5/site-packages/pygatt/backends/bgapi/bgapi.py", line 565, in expect_any
  self._packet_handlers[packet_type](response)
File "/usr/local/lib/python3.5/site-packages/pygatt/backends/bgapi/bgapi.py", line 687, in _ble_evt_gap_scan_response
  name, data_dict = self._scan_rsp_data(args['data'])
File "/usr/local/lib/python3.5/site-packages/pygatt/backends/bgapi/bgapi.py", line 506, in _scan_rsp_data
  for i in range(0, len(field_value) / 16):  # 16 bytes
TypeError: 'float' object cannot be interpreted as an integer
```

I will try and collect more debug information (the `data` list received by `BGAPIBackend._scan_rsp_data(self, data)` and the content of `field_value`).

peplin, 2016-12-08:
Interesting - the device looks like it may be returning a scan response message with a data field `complete_list_128-bit_service_class_uuids` with a number of bytes not divisible by 16. Each 128-bit service class UUID should be 16 bytes - I wonder if it's receiving partial data or there's something non-standard about the response.

danjjl, 2016-12-08:
It is most probably related to a non-standard Bluetooth device (not one we are trying to connect to) because he is now unable to reproduce the issue at the office.

danjjl, 2016-12-12:
What is your opinion on this issue?

Should `len(field_value) / 16` be surrounded by a `try`-`except` block?

peplin, 2016-12-14:
I think if the returned data is not divisible by 16, we should log a warning and ignore the packet.

PR #85 (2016-12-22): Ignore devices returning a service class UUID not dividable by 16 bytes
Fixes #82
commit a38a0f5e (2016-12-22): Ignore devices returning a service class UUID not dividable by 16 bytes (#85)

Fixes #82
closed 2016-12-22 by peplin