peplin/pygatt #83: BUG Connection not established

DavidLP, 2016-12-11:
Gatttool is working and the following lines create a connection to my device:
```
gatttool -i hci0 -I
[   ][                 ][LE]> sec-level low
[   ][                 ][LE]> connect 00:1A:22:07:B2:D1 public
[CON][00:1A:22:07:B2:D1][LE]> 
```

But a very simple example using `pygatt` that incorporates the same commands does not work:
```
import pygatt
adapter = pygatt.GATTToolBackend()
adapter.start()
device = adapter.connect("00:1A:22:07:B2:D1")
```

and gives a time out. With (enhanced) debug output the traceback is:
```
DEBUG:pygatt.backends.gatttool.gatttool:gatttool_cmd=gatttool -i hci0 -I
INFO:pygatt.backends.gatttool.gatttool:Running...
INFO:pygatt.backends.gatttool.gatttool:Connecting to 00:1A:22:07:B2:D1 with timeout=5.0
DEBUG:pygatt.backends.gatttool.gatttool:gatttool_sendline=sec-level low
DEBUG:pygatt.backends.gatttool.gatttool:gatttool_sendline=connect 00:1A:22:07:B2:D1 public
ERROR:pygatt.backends.gatttool.gatttool:Timed out connecting to 00:1A:22:07:B2:D1 after 5.0 seconds.
Traceback (most recent call last):
  File "/home/pi/.homeassistant/hass/lib/python3.5/site-packages/pygatt/backends/gatttool/gatttool.py", line 322, in connect
    self.sendline(cmd)
  File "/usr/local/lib/python3.5/contextlib.py", line 66, in __exit__
    next(self.gen)
  File "/home/pi/.homeassistant/hass/lib/python3.5/site-packages/pygatt/backends/gatttool/gatttool.py", line 163, in event
    self.wait(event, timeout)
  File "/home/pi/.homeassistant/hass/lib/python3.5/site-packages/pygatt/backends/gatttool/gatttool.py", line 137, in wait
    raise NotificationTimeout()
pygatt.exceptions.NotificationTimeout: None

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "basic_bluez_gatttool_sample.py", line 11, in <module>
    device = adapter.connect("00:1A:22:07:B2:D1")
  File "/home/pi/.homeassistant/hass/lib/python3.5/site-packages/pygatt/backends/gatttool/gatttool.py", line 328, in connect
    raise NotConnectedError(message)
pygatt.exceptions.NotConnectedError: Timed out connecting to 00:1A:22:07:B2:D1 after 5.0 seconds.
```
I am surprised, what is going wrong? My guess is that the connection state parsing [here](https://github.com/peplin/pygatt/blob/master/pygatt/backends/gatttool/gatttool.py#L96) is buggy.

DavidLP, 2016-12-18:
Is this another unmaintained BLE python library? That is sad ...

peplin, 2016-12-22:
>That is sad ...

I'm sad you're sad :(

I'm the only maintainer, and my day job has kept me busy as we wrap up 2016. I try to find time for open source every couple of weeks on the weekend.

danjjl, 2016-12-22:
Looking at the commit history and at Peplin's activity, I don't think it is fair to call this  library unmaintained!

Thank you Peplin for keeping this library alive and helping people with troubles using it :smiley: 

---
Back to your issue.

Still need to dig into the connection code for the GATTTool backend. What I can already say is that is not a Bluez 5.43 issue (I have the same version installed on my PC and can connect to devices)

peplin, 2016-12-28:
David, I looked into the BlueZ source and I have the same impression - I think you are running a different version of BlueZ. 

5.43 is the last official release from http://www.bluez.org/, and that's what I am using (in Arch Linux). It doesn't prefix the prompt with `[CON]` but prints out the `Connection successful` line. It looks like that behavior was changed in 2013 with [this commit](http://git.kernel.org/cgit/bluetooth/bluez.git/commit/?id=e80e4bd1ec7) - I'm guessing you are on a 2013 version of BlueZ.

Here's the function from BlueZ/gatttool that [builds the prompt](http://git.kernel.org/cgit/bluetooth/bluez.git/tree/attrib/interactive.c#n81) and here is the [connection callback function](http://git.kernel.org/cgit/bluetooth/bluez.git/tree/attrib/interactive.c#n154) (inside BlueZ) that prints `Connection successful`.

Granted, depending on the text output of a command line tool is a brittle way of doing things, but it's what we have to go on for now :). I would love to get this fork that add's a D-Bus backend updated and merged: https://github.com/daviessm/pygatt - then we could ditch gatttool.

Please let me know if you aren't able to update to a newer BlueZ for some reason - we may be able to patch in support for this older version if the prompt is the only required change.

closed 2016-12-28 by peplin

bluegiga1234, 2019-07-28:
I have the same error. My console output is:
Traceback (most recent call last):
  File "./basic_bluez_gatttool_sample.py", line 15, in <module>
    device = adapter.connect(YOUR_DEVICE_ADDRESS, address_type=ADDRESS_TYPE)
  File "/home/osboxes/.local/lib/python2.7/site-packages/pygatt/backends/gatttool/gatttool.py", line 421, in connect
    raise NotConnectedError(message)
pygatt.exceptions.NotConnectedError: Timed out connecting to F5:A4:29:BB:61:23 after 5.0 seconds.
I use pygatt version 4.0.3.
Any insight would be much appreciated.