peplin/pygatt #88: GATTToolBackend fails to scan twice on same adapter

danjjl, 2016-12-15:
On Ubuntu 16.04 with bluez 5.43, calling `scan()` two times on the same GATTTool adapter systematically fails.

Here is an example script of the problem :

```
>>> import pygatt
>>> adapter = pygatt.GATTToolBackend()
>>> adapter.start()
>>> adapter.scan()
[{'name': None, 'address': 'A4:77:33:C2:D0:5F'}, {'name': 'BF PPG Project', 'address': 'A0:E6:F8:75:50:84'}]
>>> adapter.scan()
Unexpected error when scanning
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/pexpect/spawnbase.py", line 150, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/pexpect/expect.py", line 99, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/local/lib/python3.5/dist-packages/pexpect/pty_spawn.py", line 465, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/local/lib/python3.5/dist-packages/pexpect/spawnbase.py", line 155, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/pygatt/backends/gatttool/gatttool.py", line 288, in scan
    scan.expect('foooooo', timeout=timeout)
  File "/usr/local/lib/python3.5/dist-packages/pexpect/spawnbase.py", line 321, in expect
    timeout, searchwindowsize, async)
  File "/usr/local/lib/python3.5/dist-packages/pexpect/spawnbase.py", line 345, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/local/lib/python3.5/dist-packages/pexpect/expect.py", line 105, in expect_loop
    return self.eof(e)
  File "/usr/local/lib/python3.5/dist-packages/pexpect/expect.py", line 50, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f4779c2d5c0>
command: /usr/bin/hcitool
args: ['/usr/bin/hcitool', 'lescan']
buffer (last 100 chars): b''
before (last 100 chars): b'Set scan parameters failed: Input/output error\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 1
flag_eof: True
pid: 8082
child_fd: 6
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile("b'foooooo'")

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/pygatt/backends/gatttool/gatttool.py", line 294, in scan
    raise BLEError(message)
pygatt.exceptions.BLEError: Unexpected error when scanning
```

danjjl, 2016-12-15:
Wrapping `scan` commands in a try-except block does solve the issue but it is more of a hack than a solutions (and kills all other ongoing Bluetooth connections)

    try:
        devices = adapter.scan(timeout=1)
    except pygatt.exceptions.BLEError:
        adapter.stop()
        adapter.start()
        devices = adapter.scan(timeout=1)

peplin, 2016-12-28:
Thanks, I was able to reproduce this error - the root cause abruptly killing `hcitool lescan`. I was able to trigger the same error by running `hcitool lescan`, then `kill -9`-ing it. PR incoming!

closed 2016-12-28 by peplin

danjjl, 2016-12-28:
Well done thank you! pulling and testing now :smiley:

PR #92 (2016-12-28): Attempt to gracefully stop lescan to leave adapter in a happy state.
If `hcitool lescan` is not allowed a graceful exit, it can leave the
adapter (or BlueZ) stack in a state where it reports a failure to set
scan parameters, and you have to reset the adapter. This change sends a
SIGINT to the process and waits for it to terminate before continuing,
allowing multiple scans without resetting.

If the run_as_root option was True, however, we can't send SIGINT as a
normal user.

Fixes #88
commit 465c8195 (2016-12-28): Attempt to gracefully stop lescan to leave adapter in a happy state. (#92)

If `hcitool lescan` is not allowed a graceful exit, it can leave the
adapter (or BlueZ) stack in a state where it reports a failure to set
scan parameters, and you have to reset the adapter. This change sends a
SIGINT to the process and waits for it to terminate before continuing,
allowing multiple scans without resetting.

If the run_as_root option was True, however, we can't send SIGINT as a
normal user.

Fixes #88