peplin/pygatt #94: autoreconnect in @connection_required decorator

peteruhnak, 2016-12-31:
Would it make sense to automatically reconnect when using the decorated methods (char_read, char_write, ...)

In an interactive environment where the user is triggering the reads/writes it is easy for the device to timeout (I would have to increase the timeout to minutes, which is too long), which means the user has to reconnect every time a read/write returns timeout and then retry the read/write.

Is there a better way to go about this? Or should I establish a brand new connection for each read/write?

Thanks!

peplin, 2017-01-18:
I am hesitant to build this functionality into pygatt itself, because it seems specific to the device or application. Is your device turning off BLE if it doesn't receive a command in a given timeout?

peteruhnak, 2017-01-18:
Well my understanding was that any BLE device should disconnect after timeout and switch back into low-energy mode. But if never had such problems, then I guess I built it incorrectly and will have to take a second look. (the board is nrf51822 btw)

peplin, 2017-04-01:
Let me know if you find anything standard in BLE about automatic disconnections - for now I'm going to consider this one closed, and let users handle auto re-connection in their own code.

closed 2017-04-01 by peplin