peplin/pygatt #100: Windows7 BGAPIBackend, ValueError: Cannot configure port.

designingSparks, 2017-01-11:
I have installed pygatt and the driver for my BLED112-v1 USB dongle from Bluegiga. The driver is Bluegiga 5.1.2600.0. The device shows up in the device manager as Bluegiga Bluetooth Low Energy (COM29). It's settings are configured by default to

- 9600bps
- 8 data bits
- 1 stop bit

If I do the following:
import pygatt
adapter = pygatt.BGAPIBackend(serial_port='COM29') #without serial_port specified, an error is thrown
adapter.start()

The following error occurs:
ValueError: Cannot configure port, some setting was wrong. Original message: WindowsError(2, 'The system cannot find the file specified.')

What could the problem be?

peplin, 2017-01-18:
Could you try connecting to the serial port with the underlying library we use, `pyserial`? That might help isolate the problem.

Grabbing one of the [pyserial exampes](http://pyserial.readthedocs.io/en/latest/shortintro.html):

```
>>> ser = serial.Serial('COM29', 9600)
>>> s = ser.read(100) 
```

designingSparks, 2017-03-29:
I was able to fix this by renaming the com port number in the driver to a value under 10.

closed 2017-03-29 by designingSparks

peplin, 2017-04-01:
> driver to a value under 10.

Very interesting - I'll mention that in a new troubleshooting section in the README.