peplin/pygatt #109: Windows 7 error in reset() in gatttool.py

designingSparks, 2017-03-29:
I'm using the development version of pygatt from github. When running the example code in basic_bluez_gatttool_sample.py, there is an error when adapter.start() is called. I have traced this to the following line in gatttool.py. The offending line is:
`subprocess.Popen(["sudo", "systemctl", "restart", "bluetooth"]).wait()`

Clearly this command is not supported on Windows.

peplin, 2017-04-01:
Hey there - unfortunately, BlueZ (where gatttool comes from) [only works on Linux](https://github.com/peplin/pygatt#pygatt---python-module-for-bluetooth-le-generic-attribute-profile-gatt). - if you want to use pygatt you'll need to use a BLE dongle that's compatible with the BGAPI backend instead.

closed 2017-04-01 by peplin

GH0st3rs, 2020-05-27:
I'm use Gentoo with OpenRC and I have the similar problem!
Why you didn't check `systemctl` or `openrc` ?