peplin/pygatt #110: Windows error due to use of pexpect in gatttool.py

designingSparks, 2017-03-29:
I'm using the development version of pygatt. In in the function start() on line 245 of gattyool.py, pexpect is used to run a command as follows:
`self._con = pexpect.spawn(gatttool_cmd, logfile=self._gatttool_logfile) `

pexpect does not work well on Windows systems although there is wexpect.
It would be good to have a cross platform solution.

designingSparks, 2017-03-29:
The fix is relatively simple. On windows, the GATTToolBackend doesn't work. If you attempt to use it as described in basic_bluez_gatttool_sample.py, you will run into the above issue.

Therefore I would suggest two enhancements:

- Meaningful error message if the user attempts to create a GATTToolBackend on Windows.
- A better example, showing how to use the BGAPIBackend on Windows.

peplin, 2017-04-01:
Unfortunately as you found, BlueZ/gatttool are only for Linux. There is an example of how to use the BGAPI backend (which works on Windows) [in the README](https://github.com/peplin/pygatt#example-use) - be aware that the BGAPI backend requires BGAPI compatible BLE dongles, which aren't the same as those that work with BlueZ.

I will definitely add an error message to save the next person some time! Thanks for the report.

PR #114 (2017-04-01): Raise an error if user tries to use GATTToolBackend in Windows.
Fixes #110
closed 2017-04-01 by peplin