peplin/pygatt #112: hcitool and gatttool no longer distributed with standard BlueZ packages

peplin, 2017-04-01:
I'm having a hard time finding any official notification, but it looks like hcitool and gatttool are being deprecated by BlueZ. The latest package installed in Arch doesn't include it, and you have to install `bluez-utils-compat` [from the AUR](https://aur.archlinux.org/packages/bluez-utils-compat/).

Ideally, pygatt can move away from shelling out the CLI tool so this doesn't matter - e.g. the work in progress in #95. 

I'm creating this issue for awareness in case anyone else is on bleeding edge distro and finds these tools are suddenly missing.

knowhy, 2017-04-02:
gattool is deprecated.

> commit b1eb2c4cd057624312e0412f6c4be000f7fc3617
> Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
> Date:   Wed Jan 4 14:23:58 2017 +0200
> 
>    build: Hide deprecated tools under --enable-deprecated
>       
>    This marks the following tools as deprecated as they are not longer
>    maintained or have been replaced by other tools:
>    
>      hciattach hciconfig hcitool hcidump rfcomm sdptool ciptool gatttool
[bluez.git](https://git.kernel.org/pub/scm/bluetooth/bluez.git/log/?qt=grep&q=gatt)

for ArchLinux you can use https://aur.archlinux.org/packages/bluez-utils-compat/ to workaround the problem. But I still think it is necessary that pygatt doesn't require an deprecated tool form bluez-utils.

knowhy, 2017-04-02:
bluepy seems to work around the problem by compiling bluez during pip install.

https://github.com/IanHarvey/bluepy/blob/master/bluepy/Makefile

luzpaz, 2017-09-06:
[image]

Croasdell, 2018-01-27:
Many thanks for shedding light on the subject.

meskarune, 2020-07-08:
You can get gatt information using

`btgatt-client -d <mac address>`

Here is a sample output I got:

```
btgatt-client -d 58:2D:34:37:00:42
Connecting to device... Done
[GATT client]# Service Added - UUID: 00001800-0000-1000-8000-00805f9b34fb start: 0x0001 end: 0x0007
[GATT client]# Service Added - UUID: 00001801-0000-1000-8000-00805f9b34fb start: 0x0008 end: 0x000b
[GATT client]# Service Added - UUID: 226c0000-6476-4566-7562-66734470666d start: 0x000c end: 0x0015
[GATT client]# Service Added - UUID: 0000180f-0000-1000-8000-00805f9b34fb start: 0x0016 end: 0x0019
[GATT client]# Service Added - UUID: 0000180a-0000-1000-8000-00805f9b34fb start: 0x001a end: 0x0024
[GATT client]# Service Added - UUID: 00001530-1212-efde-1523-785feabcd123 start: 0x0025 end: 0x002c
[GATT client]# Service Added - UUID: 0000fe95-0000-1000-8000-00805f9b34fb start: 0x002d end: 0xffff
[GATT client]# GATT discovery procedures complete
[GATT client]# 
service - start: 0x0001, end: 0x0007, type: primary, uuid: 00001800-0000-1000-8000-00805f9b34fb
	  charac - start: 0x0002, value: 0x0003, props: 0x0a, ext_props: 0x0000, uuid: 00002a00-0000-1000-8000-00805f9b34fb
	  charac - start: 0x0004, value: 0x0005, props: 0x02, ext_props: 0x0000, uuid: 00002a01-0000-1000-8000-00805f9b34fb
	  charac - start: 0x0006, value: 0x0007, props: 0x02, ext_props: 0x0000, uuid: 00002a04-0000-1000-8000-00805f9b34fb

service - start: 0x0008, end: 0x000b, type: primary, uuid: 00001801-0000-1000-8000-00805f9b34fb
	  charac - start: 0x0009, value: 0x000a, props: 0x20, ext_props: 0x0000, uuid: 00002a05-0000-1000-8000-00805f9b34fb
		  descr - handle: 0x000b, uuid: 00002902-0000-1000-8000-00805f9b34fb

service - start: 0x000c, end: 0x0015, type: primary, uuid: 226c0000-6476-4566-7562-66734470666d
	  charac - start: 0x000d, value: 0x000e, props: 0x10, ext_props: 0x0000, uuid: 226caa55-6476-4566-7562-66734470666d
		  descr - handle: 0x000f, uuid: 00002901-0000-1000-8000-00805f9b34fb
		  descr - handle: 0x0010, uuid: 00002902-0000-1000-8000-00805f9b34fb
		  descr - handle: 0x0011, uuid: 00002904-0000-1000-8000-00805f9b34fb
	  charac - start: 0x0012, value: 0x0013, props: 0x18, ext_props: 0x0000, uuid: 226cbb55-6476-4566-7562-66734470666d
		  descr - handle: 0x0014, uuid: 00002901-0000-1000-8000-00805f9b34fb
		  descr - handle: 0x0015, uuid: 00002902-0000-1000-8000-00805f9b34fb

service - start: 0x0016, end: 0x0019, type: primary, uuid: 0000180f-0000-1000-8000-00805f9b34fb
	  charac - start: 0x0017, value: 0x0018, props: 0x12, ext_props: 0x0000, uuid: 00002a19-0000-1000-8000-00805f9b34fb
		  descr - handle: 0x0019, uuid: 00002902-0000-1000-8000-00805f9b34fb

service - start: 0x001a, end: 0x0024, type: primary, uuid: 0000180a-0000-1000-8000-00805f9b34fb
	  charac - start: 0x001b, value: 0x001c, props: 0x02, ext_props: 0x0000, uuid: 00002a29-0000-1000-8000-00805f9b34fb
	  charac - start: 0x001d, value: 0x001e, props: 0x02, ext_props: 0x0000, uuid: 00002a24-0000-1000-8000-00805f9b34fb
	  charac - start: 0x001f, value: 0x0020, props: 0x02, ext_props: 0x0000, uuid: 00002a25-0000-1000-8000-00805f9b34fb
	  charac - start: 0x0021, value: 0x0022, props: 0x02, ext_props: 0x0000, uuid: 00002a27-0000-1000-8000-00805f9b34fb
	  charac - start: 0x0023, value: 0x0024, props: 0x02, ext_props: 0x0000, uuid: 00002a26-0000-1000-8000-00805f9b34fb

service - start: 0x0025, end: 0x002c, type: primary, uuid: 00001530-1212-efde-1523-785feabcd123
	  charac - start: 0x0026, value: 0x0027, props: 0x04, ext_props: 0x0000, uuid: 00001532-1212-efde-1523-785feabcd123
	  charac - start: 0x0028, value: 0x0029, props: 0x18, ext_props: 0x0000, uuid: 00001531-1212-efde-1523-785feabcd123
		  descr - handle: 0x002a, uuid: 00002902-0000-1000-8000-00805f9b34fb
	  charac - start: 0x002b, value: 0x002c, props: 0x02, ext_props: 0x0000, uuid: 00001534-1212-efde-1523-785feabcd123

service - start: 0x002d, end: 0xffff, type: primary, uuid: 0000fe95-0000-1000-8000-00805f9b34fb
	  charac - start: 0x002e, value: 0x002f, props: 0x18, ext_props: 0x0000, uuid: 00000001-0000-1000-8000-00805f9b34fb
		  descr - handle: 0x0030, uuid: 00002902-0000-1000-8000-00805f9b34fb
	  charac - start: 0x0031, value: 0x0032, props: 0x02, ext_props: 0x0000, uuid: 00000002-0000-1000-8000-00805f9b34fb
	  charac - start: 0x0033, value: 0x0034, props: 0x02, ext_props: 0x0000, uuid: 00000004-0000-1000-8000-00805f9b34fb
	  charac - start: 0x0035, value: 0x0036, props: 0x08, ext_props: 0x0000, uuid: 00000010-0000-1000-8000-00805f9b34fb
	  charac - start: 0x0037, value: 0x0038, props: 0x0a, ext_props: 0x0000, uuid: 00000013-0000-1000-8000-00805f9b34fb
	  charac - start: 0x0039, value: 0x003a, props: 0x0a, ext_props: 0x0000, uuid: 00000014-0000-1000-8000-00805f9b34fb

[GATT client]# Device disconnected: Connection reset by peer
```

There is also gattctl

```
gattctl --connect 58:2D:34:37:00:42
Connecting...
[... 6 lines cut ...]
[58:2D:34:37:00:42]    Characteristic [00000014-0000-1000-8000-00805f9b34fb]
[58:2D:34:37:00:42]    Characteristic [00000013-0000-1000-8000-00805f9b34fb]
[58:2D:34:37:00:42]    Characteristic [00000010-0000-1000-8000-00805f9b34fb]
[58:2D:34:37:00:42]    Characteristic [00000004-0000-1000-8000-00805f9b34fb]
[58:2D:34:37:00:42]    Characteristic [00000002-0000-1000-8000-00805f9b34fb]
[58:2D:34:37:00:42]    Characteristic [00000001-0000-1000-8000-00805f9b34fb]
[58:2D:34:37:00:42]  Service [00001530-1212-efde-1523-785feabcd123]
[58:2D:34:37:00:42]    Characteristic [00001534-1212-efde-1523-785feabcd123]
[... cut ...]
[58:2D:34:37:00:42]    Characteristic [00002a29-0000-1000-8000-00805f9b34fb]
[58:2D:34:37:00:42]  Service [226c0000-6476-4566-7562-66734470666d]
[58:2D:34:37:00:42]    Characteristic [226cbb55-6476-4566-7562-66734470666d]
[58:2D:34:37:00:42]    Characteristic [226caa55-6476-4566-7562-66734470666d]
[58:2D:34:37:00:42]  Service [00001801-0000-1000-8000-00805f9b34fb]
[58:2D:34:37:00:42]    Characteristic [00002a05-0000-1000-8000-00805f9b34fb]
[58:2d:34:37:00:42] Discovered, alias = TempSensor
[58:2D:34:37:00:42] Disconnected
^C
```