peplin/pygatt #118: No BGAPI compatible device detected

crobertsbmw, 2017-04-04:
I've been playing with this a lot the last couple days. At first, I wasn't having any issues with it, but now, whenever I try to call `adapter.start()` I am getting an error that No BGAPI compatible device detected.

At first it worked everytime flawlessly, then for a while, if I just kept calling adapter.start() over and over, eventually on the tenth or fifteenth time it would work. Now I can't get it to recognize the dongle at all. I've tried unplugging and plugging in the dongle, restarting my PC, nothing seems to solve the issue.

```
Traceback (most recent call last):
  File "C:\Users\user\cabelas\pygatt\backends\bgapi\bgapi.py", line 153, in _open_serial_port
    timeout=0.25)
  File "C:\Users\user\cabelas\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\user\cabelas\lib\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "C:\Users\user\cabelas\lib\site-packages\serial\serialwin32.py", line 78, in open
    self._reconfigure_port()
  File "C:\Users\user\cabelas\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    'Original message: {!r}'.format(ctypes.WinError()))
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\user\cabelas\pygatt\backends\bgapi\bgapi.py", line 192, in start
    self._open_serial_port()
  File "C:\Users\user\cabelas\pygatt\backends\bgapi\bgapi.py", line 163, in _open_serial_port
    "No BGAPI compatible device detected")
pygatt.exceptions.NotConnectedError: No BGAPI compatible device detected

```
Any help is super appreciated.

peplin, 2017-04-06:
Check out this section very recently added to the README: https://github.com/peplin/pygatt#cant-find-bgapi-device-in-windows

My guess is the COM port is incrementing every time you plug it in, and now you've got one > 9, and for whatever reason pyserial can't detect those.

crobertsbmw, 2017-04-06:
Thanks for your response!
The port is COM3. I walked through the code line by line and I found that if I put in a delay after restarting the dongle then it was able to open the COM port. In the `start()` method, I just put in a quarter second delay:
```
...
self.send_command(CommandBuilder.system_reset(0))
self._ser.flush()
self._ser.close()
time.sleep(.25)
self._open_serial_port()
...
```
This got things working for me so I guess you can close this issue if you like.

zjllynn, 2017-04-10:
Hello @crobertsbmw ,

I faced the same issue on both Linux and Windows. Could you please provide more detail on how you add a serial time delay and then solve the problem? What other library did you add? A short example code would be very helpful. 
Thank you very much!

crobertsbmw, 2017-04-10:
@zjllynn The code is posted above.  I just added the `time.sleep(.25)` line in the `start()` method. 
You are going to have to download the source and I think you need to run `python setup.py develop`. This is in place of installing via pip.
This is the file you need to edit: https://github.com/peplin/pygatt/blob/master/pygatt/backends/bgapi/bgapi.py
I didn't add any other libraries.

githubbrowser, 2017-04-12:
@zjllynn, I think you should just be able to change the file directly as it was installed by pip. When the error is printed it will show where the file is as well.

I had the same problem but had to use `time.sleep(1)` instead. For some reason 250ms was not enough.

eliquious, 2017-04-20:
I've ran into the same issue. However, I don't think a hard coded sleep is going to work well mainly due to the fact that the time required is going to vary for each machine. Alternatively, the `_open_serial_port` is supposed to poll up to 10 times anyway but it does not retry for this error.

```
    def _open_serial_port(self):
        """
        Open a connection to the named serial port, or auto-detect the first
        port matching the BLED device. This will wait until data can actually be
        read from the connection, so it will not return until the device is
        fully booted.

        Raises a NotConnectedError if the device cannot connect after 10
        attempts, with a short pause in between each attempt.
        """
        for attempt in range(MAX_RECONNECTION_ATTEMPTS):
            try:
                serial_port = self._serial_port or self._detect_device_port()
                self._ser = None

                log.debug("Attempting to connect to serial port after "
                          "restarting device")
                self._ser = serial.Serial(serial_port, baudrate=115200,
                                          timeout=0.25)
                # Wait until we can actually read from the device
                self._ser.read()
                break
            except (BGAPIError, serial.serialutil.SerialException,
                    serial_exception):
                if self._ser:
                    self._ser.close()

                # --- This block right here is the issue---
                elif attempt == 0:
                    raise NotConnectedError(
                        "No BGAPI compatible device detected")
                # ---

                self._ser = None
                time.sleep(0.25)
        else:
            raise NotConnectedError("Unable to reconnect with USB "
                                    "device after rebooting")
```

As you can see, there is already a 250ms sleep in the loop near the bottom. There is also an error reported at the end if there is no connection after 10 retries. Simply removing the `elif attempt == 0` block allows for the retry but still returns as error on failure to connect. That code block also does not comply with the description of the function (it stops the retry on the first failure). Removing the block seemed like the best solution for me because my computer varies between 2-4 retries before connecting.

githubbrowser, 2017-04-27:
@eliquious you are making a good point. I have tried the sleep on our end and it was mostly working, but would still fail every now and then. Removing the early exit from the `_open_serial_port` works fine. I have updated the pull request in https://github.com/peplin/pygatt/pull/121 to reflect this. It would be great to get this into the mainline so that we do not have to work around it locally anymore. :)

peplin, 2017-09-16:
This should be resolved by #144 - there was a bug in the auto-reconnect logic that for some reason, didn't trigger on my machine.

closed 2017-09-16 by peplin

JorgePe, 2017-09-18:
Sorry but problem still occurs with release 3.2.0

I'm using Windows 10 inside a Virtual Box VM and still need to use the workaround, adding `time.sleep(0.25)` to start() after `self.ser.close()`

I don't have a physical Windows machine to check it VirtualBox is responsible for this delay

If I specify the serial port (`serial_port='COM4'`) it still happens.

PalantirWielder, 2017-10-02:
I have Windows 7 inside of a virtual machine, and it also requires the addition of a time delay to keep the connection from failing.

EpocDotFr, 2017-10-22:
Same here on baremetal Windows 7 64 bits.

alexwhittemore, 2018-01-30:
Same problem here, on Win 10, pip version 3.2.0 and latest git, and adding @crobertsbmw's delay fixes it at least reliably enough.

franckFleet, 2018-02-20:
Same problem, but with help : https://github.com/peplin/pygatt/issues/119,
adding time.sleep(1)
between self._ser.close() and self._open_serial_port()
in C:\Users\xxxxx\AppData\Local\Programs\Python\Python36-32\Lib\site-packages\pygatt\backends\bgapi\bgapi.py (located by the errror message)

works for me.

emileb, 2018-03-22:
Delay fix above fixed it for me also

Gra55h0pper, 2018-03-29:
Same problem here. Version v3.2.0. Removing the "early exit" as per @eliquious fixed it for me.

kgracki, 2018-06-15:
Also added time.selay(1) between `self._ser.close()` and `self_open_serial_port()` in bgapi.py and it fixed my problem.

amitkrsingh08, 2018-10-05:
i had same issue adding sleep works, will try to look at a permanent fix