peplin/pygatt #119: Unable to start BLED112 dongle using adapter.start()

sbhurji, 2017-04-08:
Hi,

I have used the BLED112 dongle on a different PC and didn't face this issue but I needed to set up a new computer with pygatt and ran into issues. I was trying to execute the following code which has worked for me in the past but for some reason the 

    adapter.stop()
  File "C:\Python27\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 218, in stop
    if self._running.is_set():
AttributeError: 'NoneType' object has no attribute 'is_set'

amias-channer, 2017-05-08:
this was an issue for me on OSX Sierra (mbp with usb-c)  using pygatt installed via pip , adding this pause made it work for me.

chrisabbott, 2017-05-21:
This worked on El Capitan as well! Thanks

peplin, 2017-09-16:
This is fixed by #144 - there was a bug in the automatic connection retry logic.

closed 2017-09-16 by peplin

kevincurtiss, 2017-10-16:
Still not fixed, have latest version from repo but had to add sleep(1). On OSX

peplin, 2017-10-16:
Please provide the debug log - see the README for instructions on how to
enable it. I don't have Windows or OS X machine, and I've unfortunately
been unable to reproduce this issue in Linux. I need all of your help here!

kevincurtiss, 2017-10-16:
Interesting enough I got to work this morning and took out the sleep() trying to reproduce it and it wasn't happening. I'll try and repro and get a log for you.

kevincurtiss, 2017-10-16:
Here's a log. Interesting enough it seems somewhat related to whether my Macbook Pro is plugged into external power. Not plugged in it happens generally, plugged in it's fine.
[log.txt](https://github.com/peplin/pygatt/files/1388153/log.txt)

amias-channer, 2017-10-16:
Hello all,

My experience of this is that it is extra latency caused by the USBC
implementation on 2017 mbp.

It seems to always effect the first try and sometimes the second.

It could be related to the the fact that USBC routes power over the same
bus and this might create a delay in settling the USB links.  I would
recommend powering the laptop directly from usb and comparing results for
when it's powered via the USBC port on the adapter (mine was USBC to usb a
+ hdmi + USBC).

Cheers
Amias

kevincurtiss, 2017-10-16:
Just to clarify, I have a 2015 MBP without USB-C but there does appear to be some sort of timing issue.