peplin/pygatt #120: No BGAPI compatible device detected on Ubuntu16.04

zjllynn, 2017-04-10:
Hello,
I'm now using a BLED112 on my Ubuntu 16.04. As the instruction said, Linux should automatically recognize the device. However, I got the following errors:
```
>>> adapter = pygatt.BGAPIBackend()
>>> adapter.start()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/bgapi/bgapi.py", line 178, in start
    self._open_serial_port()
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/bgapi/bgapi.py", line 163, in _open_serial_port
    "No BGAPI compatible device detected")
pygatt.exceptions.NotConnectedError: No BGAPI compatible device detected
```
Should I specify the serial port? If so, how could I know which port the device is using?

Thanks

zjllynn, 2017-04-10:
I bought the BLED112  V1 from mouser and I tried the code with com port specification(com9) on windows laptop too. But I still got the same after I excuting `python blexTest.py` error:
`C:\Users\kl_th\Desktop\python>python bleTest.py
Traceback (most recent call last):
  File "bleTest.py", line 7, in <module>
    adapter.start()
  File "C:\Python27\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 192, in start
    self._open_serial_port()
  File "C:\Python27\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 163, in _open_serial_port
    "No BGAPI compatible device detected")
pygatt.exceptions.NotConnectedError: No BGAPI compatible device detected`

I installed python, pip and pygatt. Is there anything I should also install? Do I have to install a driver for the BLED112 for both Linux and Windows? 

Thank you very much! Look forward for your feedback soon!

peplin, 2017-09-16:
If you are still having problems, please check the README for instructions on how to collection debug logging - that will help narrow down if the device is detected and failing to connect, or not being detected at all.

closed 2017-09-16 by peplin