peplin/pygatt #129: writing >20 bytes of data

elchung, 2017-05-23:
Is it possible to write more than 20 bytes of data with a single char_write command? For some reason, when I try sending data that is greater than 20 bytes, the bgapi timesout. I know that 20 bytes is the limit for writing in one "chunk" of data for BLE, does the bgapi not automatically chop up the data into 20 byte chunks to send over, then send an "write execute" command? Thanks for any help!

Here's the logging and error I get: 

`INFO:pygatt.backends.bgapi.bgapi:Characteristic 0x00002a19-0000-1000-8000-00805f9b34fb is handle 0x35
INFO:pygatt.backends.bgapi.bgapi:Characteristic descriptor 0x00002902-0000-1000-8000-00805f9b34fb is handle 0x36
INFO:pygatt.backends.bgapi.bgapi:Characteristic 0x00002a27-0000-1000-8000-00805f9b34fb is handle 0x3f
DEBUG:pygatt.device:Looking up handle for characteristic c4f51553-e1a4-cc97-2680-1b7d70348ffb
DEBUG:pygatt.device:Found <Characteristic uuid=c4f51553-e1a4-cc97-2680-1b7d70348ffb handle=43>
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.attclient_attribute_write: 43>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.attclient_attribute_write packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<EventPacketType.attclient_procedure_completed: 17>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a EventPacketType.attclient_procedure_completed packet: Success
INFO:pygatt.device:Subscribed to uuid=c4f51553-e1a4-cc97-2680-1b7d70348ffb
DEBUG:pygatt.device:Looking up handle for characteristic c4f51553-e1a4-cc97-2680-1b7d70348ffb
DEBUG:pygatt.device:Found <Characteristic uuid=c4f51553-e1a4-cc97-2680-1b7d70348ffb handle=43>
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.attclient_attribute_write: 43>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.attclient_attribute_write packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<EventPacketType.attclient_procedure_completed: 17>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a EventPacketType.attclient_procedure_completed packet: Success
INFO:pygatt.device:Received notification on handle=0x2b, value=0xb'51000000'
43
bytearray(b'Q\x00\x00\x00')
DEBUG:pygatt.device:Looking up handle for characteristic c4f51553-e1a4-cc97-2680-1b7d70348ffb
DEBUG:pygatt.device:Found <Characteristic uuid=c4f51553-e1a4-cc97-2680-1b7d70348ffb handle=43>
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.attclient_attribute_write: 43>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a EventPacketType.attclient_attribute_value packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:attribute handle = 2b
DEBUG:pygatt.backends.bgapi.bgapi:attribute type = 1
DEBUG:pygatt.backends.bgapi.bgapi:attribute value = 0xb'51000000'
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.attclient_attribute_write packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<EventPacketType.attclient_procedure_completed: 17>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a EventPacketType.attclient_procedure_completed packet: Success
DEBUG:pygatt.device:Looking up handle for characteristic c4f51551-e1a4-cc97-2680-1b7d70348ffb
DEBUG:pygatt.device:Found <Characteristic uuid=c4f51551-e1a4-cc97-2680-1b7d70348ffb handle=38>
INFO:pygatt.device:Received notification on handle=0x2b, value=0xb'53000100'
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.attclient_attribute_write: 43>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a EventPacketType.attclient_attribute_value packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:attribute handle = 2b
DEBUG:pygatt.backends.bgapi.bgapi:attribute type = 1
DEBUG:pygatt.backends.bgapi.bgapi:attribute value = 0xb'53000100'
43
bytearray(b'S\x00\x01\x00')
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\labnet\AppData\Local\Programs\Python\Python35\lib\threading.py", line 914, in _bootstrap_inner
    self.run()
  File "C:\Users\labnet\AppData\Local\Programs\Python\Python35\lib\threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\labnet\AppData\Local\Programs\Python\Python35\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 583, in _receive
    packet_type, args = self._lib.decode_packet(packet)
TypeError: 'NoneType' object is not iterable

Traceback (most recent call last):
  File "C:/Users/labnet/Documents/Clarisonic Work Files/BLE app test/testcode.py", line 47, in <module>
    device.char_write(proto_data_uuid, data)
  File "C:\Users\labnet\AppData\Local\Programs\Python\Python35\lib\site-packages\pygatt\device.py", line 93, in char_write
    wait_for_response=wait_for_response)
  File "C:\Users\labnet\AppData\Local\Programs\Python\Python35\lib\site-packages\pygatt\backends\bgapi\device.py", line 21, in wrapper
    return func(self, *args, **kwargs)
  File "C:\Users\labnet\AppData\Local\Programs\Python\Python35\lib\site-packages\pygatt\backends\bgapi\device.py", line 114, in char_write_handle
    self._backend.expect(ResponsePacketType.attclient_attribute_write)
  File "C:\Users\labnet\AppData\Local\Programs\Python\Python35\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 521, in expect
    return self.expect_any([expected], *args, **kargs)
  File "C:\Users\labnet\AppData\Local\Programs\Python\Python35\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 557, in expect_any
    raise exc
pygatt.backends.bgapi.exceptions.ExpectedResponseTimeout: Timed out after 1.000000s waiting for [<ResponsePacketType.attclient_attribute_write: 43>]`

peplin, 2017-09-16:
Today, pygatt doesn't do any automatic chunking, so I'm not surprised that the device behaves badly if you send it more data than can fit in single char_write request. I agree, this is a simple and useful addition to the library. I think it would be best as a higher level API, so char_write would remain true to the BLE standard and expect a maximum of 20 bytes of data.

peplin, 2019-03-30:
A `char_write_long` function landed in v4.0, now released to PyPi.

closed 2019-03-30 by peplin