peplin/pygatt #130: connection interval changed by device drops connection

markirwin, 2017-05-31:
I am using a BLED112 to connect to a custom device. This device after a few seconds adjusts the connection speed and this is causing my connection to drop out. How does pygatt handle this kind of event?

peplin, 2017-09-16:
Would you mind providing a debug log of this event (see the README)? I'm curious what type of notification, if any, we get from the BGAPI device when this happens. I'm not aware of any need for userspace code to expicitly handle connection speed changes. I would assume (perhaps incorrectly) this is handled by the BLE hardware.

peplin, 2017-09-16:
A thought - is it possible the default min/max connection interval are too restrictive for your device?

From `BGAPIBackend`:
```
      def connect(self, address, timeout=5,                                                                         
                  address_type=BLEAddressType.public,                                                                                                                                                                                    
                  interval_min=60, interval_max=76, supervision_timeout=100,                                        
                  latency=0):  
```

Try adjusting interval_min and interval_max to match the interval requested by the device. The debug log should include the connection status response, including the requested connection interval.

closed 2019-03-30 by peplin