peplin/pygatt #132: How to connect to multiple devices?

JustCallMeChris, 2017-06-29:
How do I connect to multiple devices?

Are there some examples? My implementation doesn't work correctly.
My problem: if I connect to device 2 then the connection and subscribtion to device 1 lost.

My idea was something like this:
```
import pygatt

adapter = pygatt.GATTToolBackend()

try:
    adapter.start()
    device1 = adapter.connect('01:23:45:67:89:ab')
    value1 = device1.subscribe('00002b00-0000-1000-8000-00805f9b34fb', callback=callback_func)
    device2 = adapter.connect('01:27:45:99:77:cd')
    value2 = device2.subscribe('00002b00-0000-1000-8000-00805f9b34fb', callback=callback_func)
finally:
    adapter.stop()

def callback_func(handle, rawData):
    binaryData = '{}'.format(hexlify(str(rawData)))
    bD_lE = toLittleEndian(binaryData)
    queue.put(bD_lE)
```

Catpants, 2017-07-24:
I connect to multiple devices by connecting with different adapters and threads, while also using a lock so that only one device can use blue-tooth at a time. This seems to work for now, but I don't think its a long term solution.

n-bigler, 2017-07-24:
@Catpants could you give a bit more infos? Maybe a sample code?

iPsych, 2017-09-16:
I am experience same issue. 
It seems that, in windows, calling 2nd device reset's all COM ports.

uilyam, 2018-01-31:
Interesting.  Are you two still experiencing this issue @iPsych and @JustCallMeChris.  I am using code very similar to the sample code provided and not having a connection issue.  However, I do see a delay in response times for one device.  Wonder if this could be related.

peplin, 2019-03-30:
Improved connection logic landed in v4.0, now on PyPi. If this doesn't resolve your issue, please open a new ticket. Thanks!

closed 2019-03-30 by peplin