peplin/pygatt #139: Multiple device problem.

iPsych, 2017-09-07:
Whenever I open the .exe application with bgpai together, the application crashes.
I manually set the com port but the issue is not resolved.

Is there any way to peacefully use two Bluegiga dongle in one computer? Also one for compiled application and one for python bgapi scripts?

I attached two bluegiga bled112 dongle, and found only COM ports with smaller number (in this case, 'COM4') works.

Whenever I try to connect latter, below error shown. two dongles work find when connected solely, and the error on 2nd com port replicates with all possible combination of dongle attachment.

File "C:\Python27\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 165, in _open_serial_port
"No BGAPI compatible device detected")
pygatt.exceptions.NotConnectedError: No BGAPI compatible device detected

iPsych, 2017-09-12:
It is due to complicate interaction with the other app...

peplin, 2017-09-16:
My recommendation is the explicitly set the COM port for the BGAPI device you want to use with pygatt. Example from the README:

```
adapter = pygatt.BGAPIBackend(serial_port='COM9')    
```

Is it possible your .exe application is also attempting to connect to the same device?

closed 2017-09-16 by peplin

zhangzhii, 2019-06-10:
Hi, did you solve it?
I know it is an old post but I am having exactly the same problem using v4.0.3. 
The dongle with a larger port number doesn't work even when both of the dongles are idle. I did explicitly set different COM for the device in my script every time and logger shows that it is trying to connect to the right device but fail. 

```
2019-06-10 11:27:38,055 - pygatt.backends.bgapi.bgapi - DEBUG - Opening connection to serial port (attempt 10)
2019-06-10 11:27:38,055 - pygatt.backends.bgapi.bgapi - DEBUG - Failed to open serial port
Traceback (most recent call last):
  File "D:\Anaconda3_64\envs\python37\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 175, in _open_serial_port
    timeout=0.25)
  File "D:\Anaconda3_64\envs\python37\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "D:\Anaconda3_64\envs\python37\lib\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "D:\Anaconda3_64\envs\python37\lib\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM13': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
2019-06-10 11:27:38,059 - ble.device - ERROR - No BGAPI compatible device detected
```