peplin/pygatt #140: Cannot subscribe 0x0000 type characteristics or services in bgapi backend.

iPsych, 2017-09-14:
When communicating with BLE peripherals, Compiled toolkit (i.e. LightBlue) successfully connects to device and subscribes service uuid or characteristics uuid. Since the target device uses 4 characters uuid, (i.e. 3EA0), pygatt and bgapi repeatedly show error message like below.

WARNING:pygatt.backends.bgapi.bgapi:UUID 3ea3 is of unknown type
WARNING:pygatt.backends.bgapi.bgapi:UUID 3ea6 is of unknown type
WARNING:pygatt.backends.bgapi.bgapi:UUID 3e71 is of unknown type
Based on the community thread from another developer, http://www.cypress.com/forum/proc-ble/pygatt-and-cypress-ble-dongle

It seems pygatt or bgapi blocks 4-characters uuid at some point.

Anybody succeed to subscribe 4-character uuid with BLED112 dongle using pygatt or bgapi?

peplin, 2017-09-16:
Hi there! jrow sums it up well in that forum post:

>[PyGATT] does not handle a 16-bit UUID that is not an officially adopted value. The type check function returns None instead of a useful value, and the code that uses this value doesn't interpret it as a characteristic to be used. 

>This could be called either a feature or a bug, because it's actually preventing the use of characteristic UUIDs that might be invalid. Officially, all 16-bit UUIDs are reserved for entities defined by the Bluetooth SIG. All proprietary/custom entities should use 128-bit UUIDs.

Today, PyGATT ignores invalid 4 byte characteristics because those are all reserved by the BLE standard. PyGATT supports all of the valid, official 4 byte characteristic IDs.

For custom characteristics, I also recommend using a 16-byte characteristic to follow the BLE standard. Do you have control over your device's firmware, or is it from a 3rd party?

iPsych, 2017-09-16:
Unfortunately, I can't access or control the firmware. It's 3rd party device...

peplin, 2017-09-17:
I'm curious, what device is it? Regardless, I just landed a change to permit non-standard values. Give the  `master` branch a try!

closed 2017-09-17 by peplin

iPsych, 2017-09-18:
Thank you for your help.
It recognizes the uuid properly,
WARNING:pygatt.backends.bgapi.bgapi:Unrecognized 4 byte UUID 3ea3
WARNING:pygatt.backends.bgapi.bgapi:Unrecognized 4 byte UUID 3ea6

but can't passes over it to subscribe notification...

self.device.subscribe('3ea13, callback=self._handle_test) gives error 

File "/usr/local/lib/python2.7/site-packages/pygatt/device.py", line 144, in subscribe
    self._notification_handles(uuid)
  File "/usr/local/lib/python2.7/site-packages/pygatt/device.py", line 120, in _notification_handles
    value_handle = self.get_handle(uuid)
  File "/usr/local/lib/python2.7/site-packages/pygatt/device.py", line 199, in get_handle
    char_uuid = UUID(char_uuid)
  File "/usr/local/Cellar/python/2.7.13_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/uuid.py", line 136, in __init__
    raise ValueError('badly formed hexadecimal UUID string')

peplin, 2017-09-18:
Try using the full 16 byte value - the 4 byte notation is BLE shorthand for:

```
0000XXXX-0000-1000-8000-00805F9B34FB
```

where `XXXX` is replaced by those 4 bytes. In your case, try `00003EA0-0000-1000-8000-00805F9B34FB` for the first characteristic you mentioned.

iPsych, 2017-09-18:
Thank you so much! It works, and seems to working in some degree.
However, I can't write any bite to 'custom' characteristics. 
i.e. i have to write 0x4321 to 3e71, but it doesn't work at all.

listing shown me 3e71 is listed as below.
UUID('00003e71-0000-1000-8000-00805f9b34fb'): <pygatt.backends.backend.Characteristic object at 0x1091914d0>

commit 86f1312f (2017-09-17): Accept (but warn about) non-standard 4-byte characteristics.

To accommodate devices that implement custom characteristics in the
range reserved by the BLE standard, mark those as "nonstandard" but
permit them to be used.

Fixes #140
PR #148 (2017-09-17): Accept (but warn about) non-standard 4-byte characteristics.
To accommodate devices that implement custom characteristics in the
range reserved by the BLE standard, mark those as "nonstandard" but
permit them to be used.

Fixes #140
commit 79f69acc (2017-09-17): Accept (but warn about) non-standard 4-byte characteristics. (#148)

To accommodate devices that implement custom characteristics in the
range reserved by the BLE standard, mark those as "nonstandard" but
permit them to be used.

Fixes #140