peplin/pygatt #158: Can't find BGAPI device in Windows

edisonlee102, 2017-10-24:
I face a problem about find BGAPI device in Windows, i have been try the COM port of the device to a value under 10, but didn't work.

Log as below, could you give me some advices?

INFO:pygatt.backends.bgapi.bgapi:Initialized new BGAPI backend
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 1 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM2 - Bluegiga Bluetooth Low Energy (COM2)
DEBUG:pygatt.backends.bgapi.util:USB device: Bluegiga Bluetooth Low Energy (COM2) VID=0x2458 PID=0x0001 on COM2
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port COM2
INFO:pygatt.backends.bgapi.bgapi:Resetting and reconnecting to device for a clean environment
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 1 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM2 - Bluegiga Bluetooth Low Energy (COM2)
DEBUG:pygatt.backends.bgapi.util:USB device: Bluegiga Bluetooth Low Energy (COM2) VID=0x2458 PID=0x0001 on COM2
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port COM2
DEBUG:pygatt.backends.bgapi.bgapi:Failed to open serial port
Traceback (most recent call last):
  File "D:\Cypress\BleCommand_fromDongle\pygatt\pygatt\backends\bgapi\bgapi.py", line 160, in _open_serial_port
    timeout=0.25)
  File "C:\Python27\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Python27\lib\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "C:\Python27\lib\site-packages\serial\serialwin32.py", line 78, in open
    self._reconfigure_port()
  File "C:\Python27\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    'Original message: {!r}'.format(ctypes.WinError()))
SerialException: Cannot configure port, something went wrong. Original message: WindowsError(2, '???????????')
Traceback (most recent call last):
  File "Ble_Dongle_Tool.py", line 17, in <module>
    adapter.start()
  File "D:\Cypress\BleCommand_fromDongle\pygatt\pygatt\backends\bgapi\bgapi.py", line 203, in start
    self._open_serial_port()
  File "D:\Cypress\BleCommand_fromDongle\pygatt\pygatt\backends\bgapi\bgapi.py", line 171, in _open_serial_port
    "No BGAPI compatible device detected")
pygatt.exceptions.NotConnectedError: No BGAPI compatible device detected

Thank you very much.

edisonlee102, 2017-10-24:
I found that is related about self.send_command(CommandBuilder.system_reset(0)) in bgapi.py.
It will cause fail when you re-open com port.

```
        # to be plugged in.
        self._open_serial_port(max_connection_attempts=1)
        
        log.info("Resetting and reconnecting to device for a clean environment")
        # Blow everything away and start anew.
        # Only way to be sure is to burn it down and start again.
        # (Aka reset remote state machine)
        # Note: Could make this a conditional based on parameter if this
        # happens to be too slow on some systems.

        # The zero param just means we want to do a normal restart instead of
        # starting a firmware update restart.
        self.send_command(CommandBuilder.system_reset(0))
        self._ser.flush()
        self._ser.close()

```
Could you explain your ideal?

lemarquois, 2018-03-05:
I guess this is a duplication of #164

peplin, 2019-03-30:
Fixed in v4.0 (I hope!). Thanks for the report.

closed 2019-03-30 by peplin