peplin/pygatt #159: Start() fails on macOS

nremond, 2017-10-31:
Hello,

I'm using macOS High Sierra with a BLED112.
I have this very simple program that fails

```
import pygatt
import logging

logging.basicConfig()
logging.getLogger('pygatt').setLevel(logging.DEBUG)

adapter = pygatt.BGAPIBackend()
adapter.start()
```

With the following error:

```
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
INFO:pygatt.backends.bgapi.bgapi:Resetting and reconnecting to device for a clean environment
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
DEBUG:pygatt.backends.bgapi.bgapi:Failed to open serial port
Traceback (most recent call last):
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/serial/serialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
OSError: [Errno 6] Device not configured: '/dev/cu.usbmodem1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 159, in _open_serial_port
    timeout=0.25)
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 6] could not open port /dev/cu.usbmodem1: [Errno 6] Device not configured: '/dev/cu.usbmodem1'
Traceback (most recent call last):
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/serial/serialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
OSError: [Errno 6] Device not configured: '/dev/cu.usbmodem1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 159, in _open_serial_port
    timeout=0.25)
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 6] could not open port /dev/cu.usbmodem1: [Errno 6] Device not configured: '/dev/cu.usbmodem1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 202, in start
    self._open_serial_port()
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 170, in _open_serial_port
    "No BGAPI compatible device detected")
pygatt.exceptions.NotConnectedError: No BGAPI compatible device detected
>>> adapter = pygatt.BGAPIBackend()
INFO:pygatt.backends.bgapi.bgapi:Initialized new BGAPI backend
>>> adapter.start()
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 3 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: /dev/cu.Bluetooth-Incoming-Port - n/a
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: /dev/cu.n1r3-phone-WirelessiAP - n/a
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: /dev/cu.usbmodem1 - Low Energy Dongle
DEBUG:pygatt.backends.bgapi.util:USB device: Low Energy Dongle VID=0x2458 PID=0x0001 on /dev/cu.usbmodem1
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port /dev/cu.usbmodem1
INFO:pygatt.backends.bgapi.bgapi:Resetting and reconnecting to device for a clean environment
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 3 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: /dev/cu.Bluetooth-Incoming-Port - n/a
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: /dev/cu.n1r3-phone-WirelessiAP - n/a
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: /dev/cu.usbmodem1 - Low Energy Dongle
DEBUG:pygatt.backends.bgapi.util:USB device: Low Energy Dongle VID=0x2458 PID=0x0001 on /dev/cu.usbmodem1
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port /dev/cu.usbmodem1
DEBUG:pygatt.backends.bgapi.bgapi:Failed to open serial port
Traceback (most recent call last):
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/serial/serialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
OSError: [Errno 16] Resource busy: '/dev/cu.usbmodem1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 159, in _open_serial_port
    timeout=0.25)
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 16] could not open port /dev/cu.usbmodem1: [Errno 16] Resource busy: '/dev/cu.usbmodem1'
Traceback (most recent call last):
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/serial/serialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
OSError: [Errno 16] Resource busy: '/dev/cu.usbmodem1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 159, in _open_serial_port
    timeout=0.25)
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 16] could not open port /dev/cu.usbmodem1: [Errno 16] Resource busy: '/dev/cu.usbmodem1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 202, in start
    self._open_serial_port()
  File "/Users/nicolas/developments/biowatch/bw-console/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 170, in _open_serial_port
    "No BGAPI compatible device detected")
pygatt.exceptions.NotConnectedError: No BGAPI compatible device detected
```

Any idea on what I should do? 

`sudo lsof  | grep "/dev/cu.usbmodem1"` doesn't return anything :-(

thank you in advance

vinkolar, 2018-01-06:
Same problem here. Using bluegiga dongle with osX High Sierra (10.13.2). Anybody knows how to resolve the above error?

Bluegiga dongle info:
Product ID:	0x0001
Vendor ID:	0x2458
Version:	0.01

fillybrese, 2018-06-20:
In bgapi.py add time.sleep(2) after the device and serial port reset. I'm on OSX High Sierra and this works for me. Modified code below:

`    def start(self):
        """
        Connect to the USB adapter, reset it's state and start a backgroud
        receiver thread.
        """
        if self._running and self._running.is_set():
            self.stop()

        # Fail immediately if no device is attached, don't retry waiting for one
        # to be plugged in.
        self._open_serial_port(max_connection_attempts=1)

        log.info("Resetting and reconnecting to device for a clean environment")
        # Blow everything away and start anew.
        # Only way to be sure is to burn it down and start again.
        # (Aka reset remote state machine)
        # Note: Could make this a conditional based on parameter if this
        # happens to be too slow on some systems.

        # The zero param just means we want to do a normal restart instead of
        # starting a firmware update restart.

        self.send_command(CommandBuilder.system_reset(0))
        self._ser.flush()
        self._ser.close()

        time.sleep(2)

        self._open_serial_port()
        self._receiver = threading.Thread(target=self._receive)
        self._receiver.daemon = True

        self._running = threading.Event()
        self._running.set()
        self._receiver.start()
`

I'll can submit a pull request if this works for others on OSX High Sierra.

fillybrese, 2018-10-09:
Submitted Pull Request: https://github.com/peplin/pygatt/pull/201

kimpik, 2019-01-09:
I am also seeing the same issue on Windows.  The solution suggested above did resolve the delayed system_reset(0) issue. 

Is there any plan to merge this pull request to the master? 
Thanks.

peplin, 2019-03-30:
Improved connection logic landed in v4.0 (just released to PyPi). Please try the new version, and open a new issue if there are still connection issues on OS X. Thanks!

closed 2019-03-30 by peplin