peplin/pygatt #161: No characteristic found matching an UUID that is listed by bluetoothctl

alcir, 2017-11-10:
Linux environment. Using `bluetoothctl` I'm able to connect to a micro:bit device, select an attribute (Nordic UART TX), and read data, and also putting notification on in order to get data when the device write on the _UART interface_.

<pre>
[bluetooth]# connect E4:6D:B6:XX:XX:XX
Attempting to connect to E4:6D:B6:XX:XX:XX
[CHG] Device E4:6D:B6:XX:XX:XX Connected: yes
Connection successful

[BBC micro:bit]# list-attributes 
...
Primary Service
	/org/bluez/hci0/dev_E4_6D_B6_XX_XX_XX/service0021
	6e400001-b5a3-f393-e0a9-e50e24dcca9e
	Nordic UART Service
Characteristic
	/org/bluez/hci0/dev_E4_6D_B6_XX_XX_XX/service0021/char0022
	6e400002-b5a3-f393-e0a9-e50e24dcca9e
	Nordic UART TX
Descriptor
	/org/bluez/hci0/dev_E4_6D_B6_XX_XX_XX/service0021/char0022/desc0024
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
Characteristic
	/org/bluez/hci0/dev_E4_6D_B6_XX_XX_XX/service0021/char0025
	6e400003-b5a3-f393-e0a9-e50e24dcca9e
	Nordic UART RX
...

[BBC micro:bit]# select-attribute 6e400002-b5a3-f393-e0a9-e50e24dcca9e
[BBC micro:bit:/service0021/char0022]# notify on 
</pre>

Using pygatt (GATTToolBackend) I'm able to connect to the device, but the UART TX UUID is not listed, only the UART RX UUID is reported as _valid_.

<pre>
...
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value handle: 0x26
WARNING:pygatt.device:No characteristic found matching 6e400002-b5a3-f393-e0a9-e50e24dcca9e
...
</pre>

So I'm unable to use `device.subscribe` to get data as soon as it is sent by the device.

JustCallMeChris, 2017-11-19:
I had the same problem. With 2.1.1 I have no issues.

m-georgi, 2017-11-30:
I can also confirm this behaviour under version 3.2.
Under 3.1.1 all characteristics are found by pygatt.

(Bluez version 5.37-0ubuntu5.1, Ubuntu Xenial)

kalhamaar, 2017-12-12:
I can confirm with version 3.2.0
gattool can list my 3 basic services (Generic Access, Generic Attribute and Device Information and my custom one).
`[xx:xx:xx:xx:xx:xx][LE]> primary
attr handle: 0x0001, end grp handle: 0x0007 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle: 0x0008, end grp handle: 0x0008 uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle: 0x0009, end grp handle: 0x0019 uuid: 0000180a-0000-1000-8000-00805f9b34fb
attr handle: 0x001a, end grp handle: 0xffff uuid:     xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx
`
With their corresponding characteristics:
`[xx:xx:xx:xx:xx:xx][LE]> characteristics
handle: 0x0002, char properties: 0x02, char value handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0004, char properties: 0x02, char value handle: 0x0005, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0006, char properties: 0x02, char value handle: 0x0007, uuid: 00002a04-0000-1000-8000-00805f9b34fb
handle: 0x000a, char properties: 0x02, char value handle: 0x000b, uuid: 00002a25-0000-1000-8000-00805f9b34fb
handle: 0x000c, char properties: 0x02, char value handle: 0x000d, uuid: 00002a24-0000-1000-8000-00805f9b34fb
handle: 0x000e, char properties: 0x02, char value handle: 0x000f, uuid: 00002a23-0000-1000-8000-00805f9b34fb
handle: 0x0010, char properties: 0x02, char value handle: 0x0011, uuid: 00002a27-0000-1000-8000-00805f9b34fb
handle: 0x0012, char properties: 0x02, char value handle: 0x0013, uuid: 00002a26-0000-1000-8000-00805f9b34fb
handle: 0x0014, char properties: 0x02, char value handle: 0x0015, uuid: 00002a28-0000-1000-8000-00805f9b34fb
handle: 0x0016, char properties: 0x02, char value handle: 0x0017, uuid: 00002a29-0000-1000-8000-00805f9b34fb
handle: 0x0018, char properties: 0x02, char value handle: 0x0019, uuid: 00002a50-0000-1000-8000-00805f9b34fb
[...]`

But unfortunately pygatt list only two of these (Model Number String and one Custom characteristic) ;-(
`DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a24-0000-1000-8000-00805f9b34fb, value handle: 0xd
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx, value handle: 0x22`

Still investigating using debug on PyCharm, got this error that may break parsing of characteristics, if it helps:
`DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a24-0000-1000-8000-00805f9b34fb, value handle: 0xd
Traceback (most recent call last):
  File "/home/kalhamaar/softs/pycharm-community/helpers/pydev/_pydevd_bundle/pydevd_comm.py", line 1134, in do_it
    result = pydevd_vars.evaluate_expression(self.thread_id, self.frame_id, self.expression, self.doExec)
  File "/home/kalhamaar/softs/pycharm-community/helpers/pydev/_pydevd_bundle/pydevd_vars.py", line 356, in evaluate_expression
    Exec(expression, updated_globals, frame.f_locals)
  File "/home/kalhamaar/softs/pycharm-community/helpers/pydev/_pydevd_bundle/pydevd_exec.py", line 3, in Exec
    exec exp in global_vars, local_vars
  File "<string>", line 1
    self._characteristics[2a25-0000-1000-8000-00805f9b34fb]`

PS: Some data have been anonymized, thank you for understanding

deviousname, 2018-01-09:
How do I uninstall 3.2 so I can install 3.1.1?

kalhamaar, 2018-01-09:
Can use pip:
`pip install pygatt==3.1.1`

Obviously I recommend using virtualenv

FlurinArner, 2018-01-10:
I am seeing the same behavior with pygatt 3.2.0
The `_save_charecteristic_callback()` is never called, and when `discover_characteristics()` checks `if not self._characteristics:`, it fails and `Characteristic discovery failed` is raised.

With 3.1.1 the characteristics are found and no error occurs.

I ran this [MWE.txt](https://github.com/peplin/pygatt/files/1619207/MWE.txt) python script (change name to .py and enable `chmod +x MWE.py`).
This produced the two different outputs: 
* [gatttool-3.1.1-correct_output.txt](https://github.com/peplin/pygatt/files/1619212/gatttool-3.1.1-correct_output.txt)
* [gatttool-3.2.0-wrong_output.txt](https://github.com/peplin/pygatt/files/1619214/gatttool-3.2.0-wrong_output.txt).

Note: I added `log.warn("entered discover_characteristics")` and `log.warn("entered _save_charecteristic_callback")` to the file _pygatt/backends/gatttool/gatttool.py_, right after entering the different function calls.

I'm not sure why this is the case, but if someone could tell me where to look next, I'd be glad to!

dsantosp12, 2018-01-18:
Confirmed issue in 3.2, in 3.1.1 works.

IlyaSukhanov, 2018-01-19:
This is dupe of https://github.com/peplin/pygatt/issues/152

wlyzcz, 2018-11-28:
I got a strange problem,I use two characteristic to check it ,one is ok ,and another I got " No characteristic found matching"with using "subscribe",see this below
import time
import pygatt
from binascii import hexlify

adapter = pygatt.GATTToolBackend()

def handle_data(handle, value):
    print("Received data: %s" % hexlify(value))

adapter.start()
device = adapter.connect('A4:34:F1:E9:3F:5A')

while(1):
    #device.subscribe('0000fff5-0000-1000-8000-00805f9b34fb',
    #             callback=handle_data)
    device.subscribe("0000fff4-0000-1000-8000-00805f9b34fb",
                 callback=handle_data)
``
"0000fff5-0000-1000-8000-00805f9b34fb "is ok ,I got data,but I got that error with using"0000fff4-0000-1000-8000-00805f9b34fb"

peplin, 2019-03-30:
#156 landed in v4.0 and should fix this problem.

closed 2019-03-30 by peplin