peplin/pygatt #164: On Windows, No BGAPI compatible device detected

designingSparks, 2017-12-22:
This is new in version 3.2. The program gets stuck at line 163 in bgapi.py due to a serial.serialutil.SerialException. In version 3.0, this problem didn't occur.

I have a temporary fix by commenting out lines 168-170:
`elif attempt == 0:
  raise NotConnectedError(
 "No BGAPI compatible device detected")`

alxnik, 2018-02-11:
+1
BGAPIBackend connects to the device and resets it. It then tries to reconnect while the device has not reset yet so it fails. 
Adding a delay is enough to let the device reset.

lemarquois, 2018-03-05:
+1
same problem here. I also added a delay of 1s

draapho, 2018-03-13:
see here:
https://github.com/peplin/pygatt/issues/179

change one line in `_open_serial_port` function
``` python
def _open_serial_port(self, max_connection_attempts=MAX_CONNECTION_ATTEMPTS):

        for attempt in range(max_connection_attempts):
            try:
                ................................
            except (BGAPIError, serial.serialutil.SerialException, serial_exception):
                # elif attempt == 0: ######## change this line to 
                elif attempt >= max_connection_attempts:  ##### changed like this 
               ................
```

Gra55h0pper, 2018-03-29:
@draapho: I think you intended to change it to: 

    elif attempt >= max_connection_attempts-1:  ##### Note the added: '-1'

draapho, 2018-03-30:
Yes, you are right. Thank you.

draapho, 2018-03-30:
@Gra55h0pper 
Yes, you are right. Always easy to make a mistake like this. Thank you.

peplin, 2019-03-30:
This fix landed in v4.0. Thanks for the report and pull request!

closed 2019-03-30 by peplin