peplin/pygatt #166: How do I install 3.1.1 because 3.2 is not working with my device?

deviousname, 2018-01-09:
Trying to set up pygatt so I can use muse-lsl, but there is an issue and I need to install 3.1.1 correctly, can someone help? Ubuntu 16.04

FlurinArner, 2018-01-09:
Hi deviousname,
I encourage you to ask questions on [stackoverflow.com](stackoverflow.com), not here, where actual problems with the library get reported.
I'll be glad to help you out if you link your problem, but I imagine this issue will get closed soon, as isn't an issue as such.
Follow the guidelines on the [README.rst](https://github.com/peplin/pygatt/blob/master/README.rst), adjusting `pip` with `pip3`, if you're using python3.

draapho, 2018-03-15:
see here https://github.com/peplin/pygatt/issues/180.
I believe it can resolve your problem.

closed 2019-03-30 by peplin