peplin/pygatt #174: Not receiving notifications using BGAPIBackend

burnhamd, 2018-03-04:
I have followed the example for receiving notifications from a bluetooth device, but I am not able to receive any of the notifications from my device.  
My Code:
~~~PYTHON
import pygatt
from binascii import hexlify
import logging
import time

logging.basicConfig()
logging.getLogger('pygatt').setLevel(logging.DEBUG)
adapter = pygatt.BGAPIBackend()

def handle_data(handle, value):
    """
    handle -- integer, characteristic read handle the data was received on
    value -- bytearray, the data returned in the notification
    """
    print("Received data: %s" % hexlify(value))

try:
    adapter.start()
    device = adapter.connect('16:46:1A:83:19:66')

    device.subscribe("438366C1-9FAE-4B36-9575-71762012041B",
                     callback=handle_data)
    time.sleep(5)
finally:
    adapter.stop()
~~~~

My debug output is as follows:
~~~~
DEBUG:pygatt.device:Found <Characteristic uuid=438366c1-9fae-4b36-9575-71762012041b handle=32>
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.attclient_write_command: 44>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.attclient_write_command packet: Success
INFO:pygatt.device:Subscribed to uuid=438366C1-9FAE-4B36-9575-71762012041B
DEBUG:pygatt.backends.bgapi.device:Disconnecting from 16:46:1A:83:19:66
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.connection_disconnect: 29>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.connection_disconnect packet: Success
INFO:pygatt.backends.bgapi.device:Disconnected from 16:46:1A:83:19:66
INFO:pygatt.backends.bgapi.bgapi:Stopping
INFO:pygatt.backends.bgapi.bgapi:Stopping receiver
~~~~

Using the bluegiga GUI I am able to receive the notifications from the characteristic.

antoniomolram, 2018-03-19:
I have exactly the same problem!

uilyam, 2018-04-07:
Far from a python pro, but can you block the main thread with time.sleep while using subscribe?

peplin, 2019-03-30:
Please give v4.0 a try (just released). I can't tell for sure what the issue was with this code, but the new version has a lot of bug fixes. Please open a new issue if this is still a problem. Thanks!

closed 2019-03-30 by peplin