peplin/pygatt #184: Successful Subscribing to LED Matrix GATT Service but not notifications.

G-Pex, 2018-03-31:
Hi Peplin, firstly thanks for all your hard work!

I'm having a bit of an issue with the MicroBit led_matrix_state GATT service. I'm able to subscribe according to the debug log however once subscribed it never receives any notifications. I'm able to send a char_read to and read the service no problem but I need it to be running asynchronously as it's running alongside a GUI I've built.

Any suggestions?! Appreciate any pointers you could give me :)

m-menon, 2018-04-02:
Perhaps you can post some code and explain a bit more in detail. For example in some cases it may require more than just subscribing for notifications - it may also require the respective sensor to be enabled for the readings to get updated.

peplin, 2019-03-30:
Please give v4.0 a try to see if this was due to some bug in pygatt, but otherwise as @m-menon said some example code would be helpful.

closed 2019-03-30 by peplin