peplin/pygatt #185: Subscribe raises a ExpectedResponseTimeout during adapter.connect function

numancelik34, 2018-04-09:
Hi,
I have got same issue here:
It points the error like this:

File "", line 1, in 
device = adapter.connect("84:68:3e:00:ef:62", 10, pygatt.BLEAddressType.random)

File ".....pygatt\backends\bgapi\bgapi.py", line 417, in connect
raise exc

NotConnectedError

when I scanned the adapter, I can see the BLE device, however when run the above code (adapter.connect), I am having this "ExpectedResponseTimeout" error.

Could you please help me here?
Thanks
Numan

tim01121995, 2018-06-22:
Hi,

I have the same question as yours.
Have you solved this problem?

Thanks
Tim

cvaidya, 2018-06-24:
Same issue as yours. I am able to connect using gatttool on command line however and also using bluetooth-internals using chrome. So there is some issue with pygatt for sure ...

peplin, 2019-03-30:
Please give the recently released v4.0 version of the library a try, as it includes a few improvements to connection setup.

closed 2019-03-30 by peplin