peplin/pygatt #192: Pygatt does not work on Linux(Debian)

sszaharovd, 2018-05-29:
Hi. I installed a library through pip3. I'm using Python 3.6. I copied the code from the example and changed it to my MAC address. As a result, it produces an error when starting:

WARNING: No module named 'pexpect'
Traceback (most recent call last):
  File "/home/bt/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 155, in _open_serial_port
    serial_port = self._serial_port or self._detect_device_port()
  File "/home/bt/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 131, in _detect_device_port
    raise BGAPIError("Unable to auto-detect BLED112 serial port")
pygatt.backends.bgapi.exceptions.BGAPIError: Unable to auto-detect BLED112 serial port

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/bt/bt.py", line 8, in <module>
    adapter.start()
  File "/home/bt/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 187, in start
    self._open_serial_port(max_connection_attempts=1)
  File "/home/bt/venv/lib/python3.6/site-packages/pygatt/backends/bgapi/bgapi.py", line 170, in _open_serial_port
    "No BGAPI compatible device detected")
pygatt.exceptions.NotConnectedError: No BGAPI compatible device detected

For example, I create the same connection through a standard utility - a screenshot attached

[image]
[image]

How can I solve this problem?

bartkowiaktomasz, 2018-06-25:
Hi, @sszaharovd. Have you managed to solve the problem? I keep getting the same error.

sszaharovd, 2018-06-25:
Yes, I solved this problem by borrowing another class(GATTToolBackend), which I also approached.

bartkowiaktomasz, 2018-06-25:
@sszaharovd thanks. Do you have something like:

`import pygatt

adapter = pygatt.backends.BGAPIBackend()

try:
    adapter.start()
    device = adapter.connect('FF:3C:8F:22:C9:C8')
    value = device.char_read("2d30c082-f39f-4ce6-923f-3484ea480596")
finally:
    adapter.stop()`

bartkowiaktomasz, 2018-06-25:
@sszaharovd thanks. Do you have something like:
```
import pygatt

adapter = pygatt.backends.BGAPIBackend()

try:
adapter.start()
device = adapter.connect('FF:3C:8F:22:C9:C8')
value = device.char_read("2d30c082-f39f-4ce6-923f-3484ea480596")
finally:
adapter.stop()
```

sszaharovd, 2018-06-26:
class BLE:
    def __init__(self):
        self.adapter = pygatt.GATTToolBackend()

    @staticmethod
    def notify(handle, value):
        callback_value = hexlify(value)
        print('Response => ', callback_value)

    def send(self, cmd):
        try:
            self.adapter.start()
            device = self.adapter.connect(DEVICE_MAC)
            device.char_write(WRITE_DATA, bytearray(cmd), wait_for_response=True)
            device.subscribe(READ_DATA, callback=self.notify)
        except Exception as e:
            print('Error: ', format(e))
            pass

        finally:
            self.adapter.stop()

if __name__ == '__main__':
    a = BLE()
    a.send([0x00, 0x00, 0x00....])

sszaharovd, 2018-06-26:
Use pygatt.GATTToolBackend() - he will give you just what you want

bartkowiaktomasz, 2018-06-27:
@sszaharovd Thanks for your responses. Unfortunately _pygatt_ each time gave me:
 ```Timed out connecting to [MAC ADDRESS]```
I solved all the problems by using [**_bluepy_**](https://github.com/IanHarvey/bluepy)

sszaharovd, 2018-06-28:
@bartkowiaktomasz, You can also use the library bluetooth.ble. It also works fine with devices. But in this library I liked the fact that it directly works with functions gatttools, which I had to

bartkowiaktomasz, 2018-07-01:
@sszaharovd Thanks. I ended up using _pexpect_, which automates gatttool commands directly, therefore resulting in the best speed.

zhangzhii, 2018-12-27:
Hi,
I solved this problems by using a previous commit of pygatt/backends/bgapi/bgapi.py
[https://github.com/peplin/pygatt/blob/2b6c2e4007be9b63e58d51101bd8aaca46093674/pygatt/backends/bgapi/bgapi.py](url)
and replace the line 506 by:
`for i in range(0, len(field_value) // 16):  # 16 bytes`

I have no idea why the latest version of pygatt doesn't work in my case, neither on Linux nor Windows. 
Hope this will give you what you want.

peplin, 2019-03-30:
v4.0 (just released) includes a few improvements for setting up connections. I'm not sure what the last BGAPI change is referring to (the link doesn't work anymore), but give the latest version a try. Please open a new issue if there are still problems. Thanks!

closed 2019-03-30 by peplin