peplin/pygatt #193: waitForResponse = False causing char_write_handle() slowdown

erik-johnson-king, 2018-06-08:
I've got a BLE connection that is being used to send firmware to a device's boot loader. Each char_write_handle() message takes about the same amount of time all the way through the download when using waitForResponse = True. Using char_write_handle() with waitForResponse = False goes much faster, but only for a few seconds, after which is starts to bog way down. Maybe 30 or 40 writes in (with 50-byte payloads) I start to get wait times of 5 or 6 seconds per write. As far as I can tell, the time is being spent inside of the char_write_handle() call. I've tried char_write(), but that didn't change anything.

I've tried doing <my thing>._backend._receiver.clear("char_written") before each write as well, as that appears to be the only immediate difference between waitForResponse = True and waitForResponse = False, but that didn't help.

peplin, 2019-03-30:
A few fixes / improvements to the char_write logic landed in v4.0. Please give that a try, and open a new issue if you are still having problems. Thanks!

closed 2019-03-30 by peplin

foromer4, 2020-05-11:
This issue reproduces for me on version 4.0.5 , since it crashes the entire app , and hard to detect I suggest to reopen this issue. thanks, Omer