peplin/pygatt #199: Is it possible to get the RSSI using pygatt?

aeyuxi, 2018-08-09:


wstreyer, 2018-10-12:
If you are not yet connected to a device, you can do this:

```
devices = adapter.scan()
for device in devices:
    rssi = device['rssi']
    address = device['address']
    print('{} - rssi = {}'.format(address, rssi))
```

Example output:
```
D1:22:E4:F9:58:90 - rssi = -84
7A:1D:9D:C0:A2:01 - rssi = -40
28:EC:A8:29:57:F2 - rssi = -85
21:9E:EF:63:26:93 - rssi = -85
40:F2:E6:9A:C9:EB - rssi = -86
39:4C:80:D2:91:2C - rssi = -91
```

peplin, 2019-03-30:
Thanks, you found it! I'm adding mention in the README that the docs (while limited) can be found in the code itself: https://github.com/peplin/pygatt/blob/master/pygatt/device.py#L45

closed 2019-03-30 by peplin

RuiRitaYang, 2020-06-23:
Hi Will.
Thanks for the information. However, it seems like the rssi info is not provided not in the return of adapter.scan().
Do you have any idea of how I could access the rssi with PyGatt without connecting to it?

Thanks!

NOA-Charles, 2020-07-09:
> Hi Will.
> Thanks for the information. However, it seems like the rssi info is not provided not in the return of adapter.scan().
> Do you have any idea of how I could access the rssi with PyGatt without connecting to it?
> 
> Thanks!

I have the same problem, did you find a solution?

jmuellers, 2021-05-20:
It seems like the function has been removed in the mean time? How can I get rssi now?
https://github.com/peplin/pygatt/blob/master/pygatt/device.py#L53