peplin/pygatt #200: Can't write in a characteristic using the uuid wih gatttool backend but I can using gatttool directly

dBeltranGeze, 2018-10-05:
I can write a value in a characteristic using gatttool in a RPI3 but I can't using a Python3 script and pygatt library.

I have a script which gets the handle passing an uuid:
`
myBLE._device.get_handle(uuid)
`
This returns 29.

Then I try to write
`
myBLE._device.char_write(29, bytearray(listOfBytes))
`
And this returns:
No characteristic found matching 29.

If I try to write the characteristic passing the uuid, nothing happens: No error and no value written.

wstreyer, 2018-10-12:
I think the char_write() uses uuid, not handle. See the source code below for device.py
```python
def char_write(self, uuid, value, wait_for_response=False):
        """
        Writes a value to a given characteristic UUID.

        uuid -- the UUID of the characteristic to write to.
        value -- a bytearray to write to the characteristic.
        wait_for_response -- wait for response after writing.

        Example:
            my_ble_device.char_write('a1e8f5b1-696b-4e4c-87c6-69dfe0b0093b',
                                     bytearray([0x00, 0xFF]))
        """
        return self.char_write_handle(self.get_handle(uuid), value,
                                      wait_for_response=wait_for_response)

```

It seems like you want char_write_handle(), but it is not implemented in my version
```python
def char_write_handle(self, handle, value, wait_for_response=False):
        """
        Writes a value to a given characteristic handle. This can be used to
        write to the characteristic config handle for a primary characteristic.

        hande -- the handle to write to.
        value -- a bytearray to write to the characteristic.
        wait_for_response -- wait for response after writing.

        Example:
            my_ble_device.char_write(42,
                                     bytearray([0x00, 0xFF]))
        """
        raise NotImplementedError()
```

wstreyer, 2018-10-12:
Sorry, I see in the last line that you already tried the uuid.

peplin, 2019-03-30:
A few fixes for characteristic writes landed in v4.0.0. Please give it a try and open a new issue if it's not working. Thanks!

closed 2019-03-30 by peplin