peplin/pygatt #203: Backend compatibility clarification.

plainas, 2018-12-03:
pygatt's documentation says:

> Bluegiga's BGAPI, compatible with USB adapters like the BLED112.

bgapi's documentation has this in its description:

>A library for using Bluetooth Low Energy modules created by BlueGiga

Does this mean that it is compatible **only** with those devices, or **that** is also compatible with those devices.

I find this confusing. I got the " No BGAPI compatible device detected" error when using this backend, but I am not sure if it's the bug reported ina couple of other issues, or if it is my Bluetooth device that is not compatible.
I am using raspberry pi 3 B+' built in Bluetooth chip.

wstreyer, 2018-12-13:
You need this dongle exactly: BLED112

https://www.silabs.com/products/wireless/bluetooth/bluetooth-low-energy-modules/bled112-bluetooth-smart-dongle

I would recommend ordering it from Silicon Labs directly, but you can probably find it on Amazon as well. Other dongles might be compatible, but I haven't used them. A standard Bluetooth dongle is not compatible since Windows 10 does not natively support BLE. It only works well with Bluetooth Classic.

The BLED112 sets up a link between a serial port and a BLE device. This library communicates with the BLED112. The BLED112 is referred to as the 'adapter' in this library. The BLE device it connects with is the 'device'. Once connected, you can query the device services and subscribe to the one you want.

peplin, 2019-03-30:
Thanks, @wstreyer - that's correct. The BGAPI (by Bluegiga) is written to be generic but to date they only have implemented it with one adapter, the BLED112 (as far as I'm aware).

closed 2019-03-30 by peplin