peplin/pygatt #205: A little help for the confused ?  :)

cinoaz, 2018-12-13:
I'm new to Python, new to BLE, I'm swimming in an ocean.  I'm learning but it just seems a bit too much shotgun approach.  Maybe someone can help with a series of links on how to get started.

1. Does a standard Bluetooth adapter on my Windows 10 laptop work with Pygatt or do I need a specific USB device?
2. Are there a series of links, videos, or whatever to help me learn Pygatt?  I have done programming in the past but not so much "From scratch".  I usually start from a known working point and add/remove/update from there.  I'm not adverse to learning, given tho I'm closer to 50...then say 40....but this BLE interface is new and anything new is daunting.

Any guidance is appreciated !

peplin, 2019-03-30:
Thanks @wstreyer! We recently merged another sample program that might help you get started, but if you are brand new to BLE you probably want to dig around for general resources on the web. I won't claim that pygatt is good for beginners.

closed 2019-03-30 by peplin

WarishaAli, 2020-04-02:
@wstreyer   can I connect my laptop (central) and android device (peripheral) using pygatt. Asking this because you said:

> Your phone is typically a central, not a peripheral, device. It is usually not advertising.

If I make an app on android that will be advertising, and then I'd be able to do the scan and connection using pygatt?

CLDiego, 2020-04-02:
@WarishaAli In theory, yes. If you're using Windows 10 you'll need the ble112 dongle to use your PC as a central and connect to your peripheral.

As @wstreyer said, usually the phone is used as a central device. I haven't seen many examples of using a phone as a BLE server. If you're planning to use your phone to communicate with your laptop through BLE, I think there are better and easier ways to do this. For example, using traditional Bluetooth or WiFi.

If you just want to learn BLE it is worth investing in a small BLE device that you can easily program as a peripheral and then connect through pygatt. For instance, esp32 has a ble library and it's an inexpensive piece of hardware.

That being said, I think that could be the best starting point to learn BLE, creating your own peripheral on a small piece of hardware like the esp32. Good luck!