peplin/pygatt #208: BGAPI fails after finding BLED112  when flushing before reopening

rogermorrell, 2019-03-14:
Windows 10 and Python 3.6

This program
import pygatt
from binascii import hexlify
import logging
import time

logging.basicConfig(filename="pg.log")
logging.getLogger('pygatt').setLevel(logging.DEBUG)

 #adapter = pygatt.BGAPIBackend(serial_port='COM16')
adapter = pygatt.BGAPIBackend()

try:
    adapter.start()
    
    device = adapter.connect('01:23:45:67:89:ab')
    value = device.char_read("a1e8f5b1-696b-4e4c-87c6-69dfe0b0093b")
finally:
    adapter.stop(

fails in adapter_start()
This is the log file
[pg.log](https://github.com/peplin/pygatt/files/2967034/pg.log)

This is the end of the log file

DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 4 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM5 - Standard Serial over Bluetooth link (COM5)
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM16 - Bluegiga Bluetooth Low Energy (COM16)
DEBUG:pygatt.backends.bgapi.util:USB device: Bluegiga Bluetooth Low Energy (COM16) VID=0x2458 PID=0x0001 on COM16
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM4 - Standard Serial over Bluetooth link (COM4)
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM8 - Standard Serial over Bluetooth link (COM8)
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port COM16
INFO:pygatt.backends.bgapi.bgapi:Resetting and reconnecting to device for a clean environment

and  as part of this it tries to flush the port and fails
 and stepping through  i get to where it is getting from windows what it failed and it gives 

code = 22
descr = The device does not recognize the command

    def WinError(code=None, descr=None):
        if code is None:
            code = GetLastError()
        if descr is None:
            descr = FormatError(code).strip()
        return OSError(None, descr, None, code)

But earlier in the log file I get 

what appears to a failure to open the BLED 112 

DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM5 - Standard Serial over Bluetooth link (COM5)
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM16 - Bluegiga Bluetooth Low Energy (COM16)
DEBUG:pygatt.backends.bgapi.util:USB device: Bluegiga Bluetooth Low Energy (COM16) VID=0x2458 PID=0x0001 on COM16
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM4 - Standard Serial over Bluetooth link (COM4)
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM8 - Standard Serial over Bluetooth link (COM8)
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port COM16
DEBUG:pygatt.backends.bgapi.bgapi:Failed to open serial port
Traceback (most recent call last):
  File "C:\Program Files\Python36\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 159, in _open_serial_port
    timeout=0.25)
  File "C:\Program Files\Python36\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Program Files\Python36\lib\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "C:\Program Files\Python36\lib\site-packages\serial\serialwin32.py", line 78, in open
    self._reconfigure_port()
  File "C:\Program Files\Python36\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    'Original message: {!r}'.format(ctypes.WinError()))
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
INFO:pygatt.backends.bgapi.bgapi:Initialized new BGAPI backend

This that traceback  from the flush failure of something else ? a background task ?  I stepped through the program up to the Resetting and reconnecting at the end of the log file and  and did not see

mivola, 2019-03-15:
I have the same/similar problem and it is probably related to #158, #164 . The fix is already here but not yet merged:  #181

rogermorrell, 2019-03-15:
After the initial post I read some more of the issue and found what mivola mentions
So I noticed that was a version with the correction ( except it was missing the -1) done by VernierST so I got that and tried it .  I no longer get that error 

But now it fails on device = adapter.connect('01:23:45:67:89:ab')  it appears to connect but them fail.  What I do not understand is that I get a different (better!) result when running the test program from the command line - compared when I run it from inside PyCharm's IDE

I'm trying to connect a Microchip RN4871 on one their dev board s where I set the 0123456789ab

peplin, 2019-03-30:
Improved device detection logic landed in v4.0.0, which was just released to PyPi. Please open a new issue if this does not resolve the problem. Thanks!

closed 2019-03-30 by peplin

GianlucaLoi, 2019-12-04:
Hello @peplin,

I'm still having this problem. 
I put some logs here:

INFO:pygatt.backends.bgapi.bgapi:Resetting and reconnecting to device for a clean environment
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
DEBUG:pygatt.backends.bgapi.bgapi:Failed to open serial port
Traceback (most recent call last):
  File "C:\Users\Gianluca.Loi\Desktop\GIT_R&D_B810\uBox_BLE_Test\venv\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 175, in _open_serial_port
    timeout=0.25)
  File "C:\Users\Gianluca.Loi\Desktop\GIT_R&D_B810\uBox_BLE_Test\venv\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\Gianluca.Loi\Desktop\GIT_R&D_B810\uBox_BLE_Test\venv\lib\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "C:\Users\Gianluca.Loi\Desktop\GIT_R&D_B810\uBox_BLE_Test\venv\lib\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM10': FileNotFoundError(2, 'Impossibile trovare il file specificato.', None, 2)
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 2)
DEBUG:pygatt.backends.bgapi.bgapi:Failed to open serial port
Traceback (most recent call last):
  File "C:\Users\Gianluca.Loi\Desktop\GIT_R&D_B810\uBox_BLE_Test\venv\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 175, in _open_serial_port
    timeout=0.25)
  File "C:\Users\Gianluca.Loi\Desktop\GIT_R&D_B810\uBox_BLE_Test\venv\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\Gianluca.Loi\Desktop\GIT_R&D_B810\uBox_BLE_Test\venv\lib\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "C:\Users\Gianluca.Loi\Desktop\GIT_R&D_B810\uBox_BLE_Test\venv\lib\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM10': FileNotFoundError(2, 'Impossibile trovare il file specificato.', None, 2)
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 3)
INFO:pygatt.backends.bgapi.bgapi:Running receiver
INFO:pygatt.backends.bgapi.bgapi:Disabling advertising
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_set_mode: 58>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_set_mode packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.sm_set_bondable_mode: 51>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.sm_set_bondable_mode packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Stopping any outstanding GAP procedure
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_end_procedure: 61>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_end_procedure packet: Device in wrong state