peplin/pygatt #229: pygatt looking for new maintainers

peplin, 2019-03-30:
Do you enjoy using pygatt? Do you have a library that depends on it?

I'm looking for additional folks to add as official maintainers of the project. My needs for a Python BLE library have diminished over the last few years, making it harder to find the time to take care of this project. 

Please let me know if you are interested! The tests and PyPi publishing are entirely automated, so it's very low overhead. You just need to be able to test changes with real hardware before merging.

nothead31, 2021-07-01:
I have a couple of things that I could test this tool on and would be interested in becoming the maintainer, if you'd like.

peplin, 2021-07-02:
Great! Give it a go with your devices, and send any fixes or improvements you have through as pull requests. If it proves to be a good match I can get you added as an owner.