peplin/pygatt #232: pygatt times out connecting to device -RPi

Thilroy, 2019-04-02:
Hi,
I've read a few posts already mentionning the same issue and haven't seen a satisfying answer yet.
My setup is as follows : Rasperry Pi3B with raspian Stretch. Python is on 3.5.3.
Bluez version is on 5.43
I've installed the latest pygatt by doing a "pip3 install pygatt" and also the "pygatt[GATTOOL]" stuff.
I'm trying to connect to a bicycle speed sensor, which I can achieve easily through bluetoothctl and gatttool altogether. Notifications work fine as well.

But when I'm trying the simple code :

import pygatt
from binascii import hexlify
MAC1='CE:1D:E1:2C:86:73'
adapter = pygatt.GATTToolBackend()
def handle_data(handle, value):
    """
    handle -- integer, characteristic read handle the data was received on
    value -- bytearray, the data returned in the notification
    """
    print("Received data: %s" % hexlify(value))

try:
    adapter.start()
    device = adapter.connect(MAC1)
    device.subscribe("00002a5b-0000-1000-8000-00805f9b34fb",callback=handle_data)
finally:
    adapter.stop()

I get the same exception : 

  File "<ipython-input-29-630533537ddd>", line 1, in <module>
    runfile('/home/pi/Documents/CO2/pygatt1.py', wdir='/home/pi/Documents/CO2')
  File "/usr/lib/python3/dist-packages/spyder/utils/site/sitecustomize.py", line 866, in runfile
    execfile(filename, namespace)
  File "/usr/lib/python3/dist-packages/spyder/utils/site/sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)
  File "/home/pi/Documents/CO2/pygatt1.py", line 26, in <module>
    device = adapter.connect(MAC1)
  File "/usr/local/lib/python3.5/dist-packages/pygatt/backends/gatttool/gatttool.py", line 420, in connect
    raise NotConnectedError(message)
NotConnectedError: Timed out connecting to CE:1D:E1:2C:86:73 after 5.0 seconds.

What could possibly go wrong here ?

Thanks in advance for your help,

Thierry

peplin, 2019-04-02:
Hi Therry, I wonder if your device uses 'random' addressing.

Try this:

```
device = adapter.connect(MAC1, address_type=pygatt.BLEAddressType.random)
```

https://github.com/peplin/pygatt/blob/master/samples/basic_bluez_gatttool_sample.py#L10

Thilroy, 2019-04-02:
Thanks, I ran onto that one too with bluepy. I didn't know how to pass the info "random" through the connect method.
Indeed it works now, thanks a lot ! But now I'm running on another message, I don't know if it's hardware-related, but I'm getting a "pygatt.exceptions.NotConnectedError : Characteristic discovery failed". I double-checked it and it's the right handle for the char, though. It's the one I get notifications from, containing the speed and timeticks data....
Am I doing something wrong ?

PR #311 (2021-02-21): Mention random addressing in README
Fixes #232
commit 2de42eaf (2021-02-21): Mention random addressing in README

Fixes #232
closed 2021-02-21 by peplin