peplin/pygatt #240: Failing to connect to device using BGAPIBackend

andreaortuno, 2019-05-01:
Hi, I'm trying to use this library instead of going through bglib to use the BLED112. However, I'm getting an error when I try to connect to my device using adapter.connet(). This does not happen when I use bglib only. Any insights would be appreciated. Also I'm using the latest development version pygatt==4.0.3.

Here is my code:
```python
import pygatt
import logging

logging.basicConfig()
logging.getLogger('pygatt').setLevel(logging.DEBUG)

adapter = pygatt.BGAPIBackend(serial_port='COM6')

BLE_CHAR_RECEIVE = "2d30c082f39f4ce6923f3484ea480596"
BLE_CHAR_SEND = "2d30c083f39f4ce6923f3484ea480596"

def handle_data(handle, value):
    print(value)

adapter.start()
device = adapter.connect('e6:5d:54:f2:f4:38')
device.bond()
```

and here is the terminal output:
```

DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 3)
INFO:pygatt.backends.bgapi.bgapi:Running receiver
INFO:pygatt.backends.bgapi.bgapi:Disabling advertising
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_set_mode: 58>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_set_mode packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.sm_set_bondable_mode: 51>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.sm_set_bondable_mode packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Stopping any outstanding GAP procedure
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_end_procedure: 61>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_end_procedure packet: Device in wrong state
INFO:pygatt.backends.bgapi.bgapi:Connecting to device at address e6:5d:54:f2:f4:38 (timeout 5s)
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.sm_set_bondable_mode: 51>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.sm_set_bondable_mode packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_connect_direct: 60>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_connect_direct packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<EventPacketType.connection_status: 11>] within 5.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_end_procedure: 61>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_end_procedure packet: Success
Traceback (most recent call last):
  File "ganglion.py", line 16, in <module>
    device = adapter.connect('e6:5d:54:f2:f4:38')
  File "C:\Users\USER\Documents\Environments\py3\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 456, in connect
    raise exc
pygatt.exceptions.NotConnectedError
```

Stimaleger, 2019-05-20:
try this : 
``` python
device = adapter.connect(address='e6:5d:54:f2:f4:38', address_type=pygatt.BLEAddressType.random)
```

andreaortuno, 2019-05-20:
Solved the issue! thanks!

closed 2019-05-20 by andreaortuno