peplin/pygatt #243: Timed out after 1.000000s / Packet class 110 and command 128 is not recognized

tihoulas, 2019-05-24:
Hi, running a test scan under ubuntu 19.04, python 2.7.16 always fails. Here is the debug log :+1: 

` INFO:pygatt.backends.bgapi.bgapi:Initialized new BGAPI backend
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 1 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: /dev/ttyACM0 - Low Energy Dongle - CDC control
DEBUG:pygatt.backends.bgapi.util:USB device: Low Energy Dongle - CDC control VID=0x2458 PID=0x0001 on /dev/ttyACM0
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port /dev/ttyACM0
INFO:pygatt.backends.bgapi.bgapi:Resetting and reconnecting to device for a clean environment
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 1 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: /dev/ttyACM0 - Low Energy Dongle - CDC control
DEBUG:pygatt.backends.bgapi.util:USB device: Low Energy Dongle - CDC control VID=0x2458 PID=0x0001 on /dev/ttyACM0
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port /dev/ttyACM0
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 0 serial USB devices
DEBUG:pygatt.backends.bgapi.bgapi:Failed to open serial port
Traceback (most recent call last):
  File "/home/tihoulas/.local/lib/python2.7/site-packages/pygatt/backends/bgapi/bgapi.py", line 171, in _open_serial_port
    serial_port = self._serial_port or self._detect_device_port()
  File "/home/tihoulas/.local/lib/python2.7/site-packages/pygatt/backends/bgapi/bgapi.py", line 147, in _detect_device_port
    raise BGAPIError("Unable to auto-detect BLED112 serial port")
BGAPIError: Unable to auto-detect BLED112 serial port
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 2)
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 0 serial USB devices
DEBUG:pygatt.backends.bgapi.bgapi:Failed to open serial port
Traceback (most recent call last):
  File "/home/tihoulas/.local/lib/python2.7/site-packages/pygatt/backends/bgapi/bgapi.py", line 171, in _open_serial_port
    serial_port = self._serial_port or self._detect_device_port()
  File "/home/tihoulas/.local/lib/python2.7/site-packages/pygatt/backends/bgapi/bgapi.py", line 147, in _detect_device_port
    raise BGAPIError("Unable to auto-detect BLED112 serial port")
BGAPIError: Unable to auto-detect BLED112 serial port
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 3)
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 1 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: /dev/ttyACM0 - Low Energy Dongle - CDC control
DEBUG:pygatt.backends.bgapi.util:USB device: Low Energy Dongle - CDC control VID=0x2458 PID=0x0001 on /dev/ttyACM0
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port /dev/ttyACM0
INFO:pygatt.backends.bgapi.bgapi:Running receiver
INFO:pygatt.backends.bgapi.bgapi:Disabling advertising
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_set_mode: 58>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_set_mode packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.sm_set_bondable_mode: 51>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.sm_set_bondable_mode packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Stopping any outstanding GAP procedure
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_end_procedure: 61>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_end_procedure packet: Device in wrong state
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_set_scan_parameters: 64>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_set_scan_parameters packet: Success
INFO:pygatt.backends.bgapi.bgapi:Starting an active scan
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_discover: 59>] within 1.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_discover packet: Success
INFO:pygatt.backends.bgapi.bgapi:Pausing for maximum 3s to allow scan to complete
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<EventPacketType.gap_scan_response: 28>] within 3.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a EventPacketType.gap_scan_response packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Received a scan response from DB:61:A4:89:CE:0C with rssi=-79 dBM and data={'complete_local_name': u'roulirouli\x00\x00\x00', 'manufacturer_specific_data': bytearray(b'NB!\x00\x00\x00\x00\xde')}
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<EventPacketType.gap_scan_response: 28>] within 3.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a EventPacketType.gap_scan_response packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Received a scan response from DB:61:A4:89:CE:0C with rssi=-79 dBM and data={'complete_list_128-bit_service_class_uuids': [u'0x6E:40:00:01:B5:A3:F3:93:E0:A9:E5:0E:24:DC:CA:9E']}
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<EventPacketType.gap_scan_response: 28>] within 3.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a EventPacketType.gap_scan_response packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Received a scan response from DB:61:A4:89:CE:0C with rssi=-73 dBM and data={'complete_local_name': u'roulirouli\x00\x00\x00', 'manufacturer_specific_data': bytearray(b'NB!\x00\x00\x00\x00\xde')}
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<EventPacketType.gap_scan_response: 28>] within 3.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a EventPacketType.gap_scan_response packet: Success
DEBUG:pygatt.backends.bgapi.bgapi:Received a scan response from DB:61:A4:89:CE:0C with rssi=-73 dBM and data={'complete_list_128-bit_service_class_uuids': [u'0x6E:40:00:01:B5:A3:F3:93:E0:A9:E5:0E:24:DC:CA:9E']}
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<EventPacketType.gap_scan_response: 28>] within 3.000000s
DEBUG:pygatt.backends.bgapi.bgapi:Received a EventPacketType.gap_scan_response packet: Success
[... 148 lines cut ...]
UnknownMessageType: Packet class 110 and command 128 is not recognized
WARNING:pygatt.backends.bgapi.bgapi:Ignoring message decode failure
Traceback (most recent call last):
  File "/home/tihoulas/.local/lib/python2.7/site-packages/pygatt/backends/bgapi/bgapi.py", line 645, in _receive
    packet_type, args = self._lib.decode_packet(packet)
  File "/home/tihoulas/.local/lib/python2.7/site-packages/pygatt/backends/bgapi/bglib.py", line 877, in decode_packet
    packet_class, packet_command, payload, payload_length)
  File "/home/tihoulas/.local/lib/python2.7/site-packages/pygatt/backends/bgapi/bglib.py", line 379, in _decode_response_packet
    (packet_class, packet_command))
UnknownMessageType: Packet class 110 and command 128 is not recognized
INFO:pygatt.backends.bgapi.bgapi:Stopping scan
DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_end_procedure: 61>] within 1.000000s
INFO:pygatt.backends.bgapi.bgapi:Stopping
INFO:pygatt.backends.bgapi.bgapi:Stopping receiver
Traceback (most recent call last):
  File "test.py", line 11, in <module>
    adapter.scan(run_as_root=True, timeout=3)
  File "/home/tihoulas/.local/lib/python2.7/site-packages/pygatt/backends/bgapi/bgapi.py", line 368, in scan
    self.expect(ResponsePacketType.gap_end_procedure)
  File "/home/tihoulas/.local/lib/python2.7/site-packages/pygatt/backends/bgapi/bgapi.py", line 577, in expect
    return self.expect_any([expected], *args, **kargs)
  File "/home/tihoulas/.local/lib/python2.7/site-packages/pygatt/backends/bgapi/bgapi.py", line 613, in expect_any
    raise exc
pygatt.backends.bgapi.exceptions.ExpectedResponseTimeout: Timed out after 1.000000s waiting for [<ResponsePacketType.gap_end_procedure: 61>]
`

gsajko, 2020-02-09:
I have the same problem, using Python3. 
`pygatt.backends.bgapi.bglib.UnknownMessageType: Packet class 255 and command 27 is not recognized`

peplin, 2021-02-21:
Unfortunately I can't reproduce this, and I can't find any definitions for BGAPI response types with those values. It's possible the device is returning corrupt data, or there's a new firmware version or device with a different API. PyGATT's BGAPI is built against this spec: https://www.silabs.com/documents/public/reference-manuals/Bluetooth_Smart_Software-BLE-1.4-API-RM.pdf

closed 2021-02-21 by peplin