peplin/pygatt #251: Failing to connect while using BGAPIBackend

bluegiga1234, 2019-07-29:
Hi Peplin,
Thank you for active work regarding Pygatt.
I'm trying to use this library instead of going through bglib to use the BLED112. However, I'm getting an error when I run the sample program, "subscribe_indicate_theremometer_sample.py". I used BGAPIBackend instead of GATTToolBackend and I ran in Windows 10 with python 3.7.4. I am using the latest development version pygatt 4.0.3. Any insights would be appreciated.
Here is the sample code that I am having problems with:

    adapter = pygatt.BGAPIBackend()

    try:
    # Start the adapter
    adapter.start()
    # Connect to the device with that given parameter.
    # For scanning, use adapter.scan()
    device = adapter.connect("F5:A4:29:BB:61:23")
    # Set the security level to medium
    device.bond()
    # Observes the given characteristics for indications.
    # When a response is available, calls data_handle_cb
    device.subscribe("00002a1c-0000-1000-8000-00805f9b34fb",
                     callback=data_handler_cb,
                     indication=True)

    input("Press enter to stop program...\n")

    finally:
    # Stop the adapter session
    adapter.stop()

    return 0

and here is the terminal output:

      INFO:pygatt.backends.bgapi.bgapi:Initialized new BGAPI backend
      INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
      DEBUG:pygatt.backends.bgapi.util:Found 4 serial USB devices
      DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM1 - Communications Port (COM1)
      DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM9 - Bluegiga Bluetooth Low Energy (COM9)
      DEBUG:pygatt.backends.bgapi.util:USB device: Bluegiga Bluetooth Low Energy (COM9)       VID=0x2458 PID=0x0001 on COM9
      DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM10 - JLink CDC UART Port (COM10)
      DEBUG:pygatt.backends.bgapi.util:USB device: JLink CDC UART Port (COM10) VID=0x1366 PID=0x1015 on COM10
      DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM12 - Bluegiga Bluetooth Low Energy (COM12)
      DEBUG:pygatt.backends.bgapi.util:USB device: Bluegiga Bluetooth Low Energy (COM12) VID=0x2458 PID=0x0001 on COM12
      INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port COM9
      INFO:pygatt.backends.bgapi.bgapi:Resetting and reconnecting to device for a clean environment
      DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
      INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
      DEBUG:pygatt.backends.bgapi.util:Found 4 serial USB devices
      DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM1 - Communications Port (COM1)
      DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM9 - Bluegiga Bluetooth Low Energy (COM9)
      DEBUG:pygatt.backends.bgapi.util:USB device: Bluegiga Bluetooth Low Energy (COM9) VID=0x2458 PID=0x0001 on COM9
      DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM10 - JLink CDC UART Port (COM10)
      DEBUG:pygatt.backends.bgapi.util:USB device: JLink CDC UART Port (COM10) VID=0x1366 PID=0x1015 on COM10
      DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM12 - Bluegiga Bluetooth Low Energy (COM12)
      DEBUG:pygatt.backends.bgapi.util:USB device: Bluegiga Bluetooth Low Energy (COM12) VID=0x2458 PID=0x0001 on COM12
      INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port COM9
      DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
      INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
      DEBUG:pygatt.backends.bgapi.util:Found 3 serial USB devices
      DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM1 - Communications Port (COM1)
      DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM10 - JLink CDC UART Port (COM10)
      DEBUG:pygatt.backends.bgapi.util:USB device: JLink CDC UART Port (COM10) VID=0x1366 PID=0x1015 on COM10
      DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM12 - Bluegiga Bluetooth Low Energy (COM12)
      DEBUG:pygatt.backends.bgapi.util:USB device: Bluegiga Bluetooth Low Energy (COM12) VID=0x2458 PID=0x0001 on COM12
      INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port COM12
      INFO:pygatt.backends.bgapi.bgapi:Running receiver
      INFO:pygatt.backends.bgapi.bgapi:Disabling advertising
      DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_set_mode: 58>] within 1.000000s
      DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_set_mode packet: Success
      DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.sm_set_bondable_mode: 51>] within 1.000000s
      DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.sm_set_bondable_mode packet: Success
      DEBUG:pygatt.backends.bgapi.bgapi:Stopping any outstanding GAP procedure
      DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_end_procedure: 61>] within 1.000000s
      DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_end_procedure packet: Device in wrong state
      INFO:pygatt.backends.bgapi.bgapi:Connecting to device at address F5:A4:29:BB:61:23 (timeout 5s)
      DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.sm_set_bondable_mode: 51>] within 1.000000s
      DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.sm_set_bondable_mode packet: Success
      DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_connect_direct: 60>] within 1.000000s
      DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_connect_direct packet: Success
      DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<EventPacketType.connection_status: 11>] within 5.000000s
      DEBUG:pygatt.backends.bgapi.bgapi:Expecting a response of one of [<ResponsePacketType.gap_end_procedure: 61>] within 1.000000s
      DEBUG:pygatt.backends.bgapi.bgapi:Received a ResponsePacketType.gap_end_procedure packet: Success
      INFO:pygatt.backends.bgapi.bgapi:Stopping
      INFO:pygatt.backends.bgapi.bgapi:Stopping receiver
      Traceback (most recent call last):
       File "subscribe_indicate_thermometer_sample.py", line 63, in <module>
      exit(main())
       File "subscribe_indicate_thermometer_sample.py", line 44, in main
       device = adapter.connect("F5:A4:29:BB:61:23")
       File "C:\Users\tchinna\AppData\Local\Programs\Python\Python37\lib\site-      packages\pygatt\backends\bgapi\bgapi.py", line 456, in connect
       raise exc
      pygatt.exceptions.NotConnectedError
        return 0

peplin, 2021-02-21:
Sorry I can't be of much help here. The device looks like it's closing the connection.

closed 2021-02-21 by peplin

A-Ferjani, 2024-04-23:
Hi there,
did you make some progress regarding the use of pygatt bgapi without an adapter on windows ? is it possible ?